ExpressLRS/ExpressLRS #1797: AUTO_WIFI_ON_INTERVAL defaults to 60s when set to off

paveltar, 2022-08-25:
Not sure if the report more suitable for this repo or configurator's one but still...
## Current Behavior

Removing checkbox for AUTO_WIFI_ON_INTERVAL flag before flash still makes the receiver go to WIFI mode after 60 seconds.
Setting the flag via receivers web interface is still possible (0 value enters the WIFI mode immediately after power on).
Was tested on HM EP1 & MATEK R24D receivers.

## Steps to Reproduce

1. Flash HM EP1 or MATEK R24D receiver with AUTO_WIFI_ON_INTERVAL flag turned off.
2. Wait for 60 seconds -> the receiver will go to WIFI mode

## Possible Solution (Not obligatory)

Seems like 0 value should disable the option?

## Details

Happens on 3.0 RC2 version
Might be a frustrating in race leaving the quad turned on before the heat starts, realizing that the quad went into WIFI mode

## Your Environment

- TX hardware: - 
- RX hardware: HM EP1 or MATEK R24D
- Handset model: -
- OpenTX version (including nightly number): -
- ExpressLRS version (TX & RX MUST MATCH): 3.0 RC2
- Packet Rate: -
- Telemetry Ratio: -
- user_defines: -

pkendall64, 2022-08-25:
-1 is disabled

pfeerick, 2022-09-14:
FYI, issue is still open as right keyword wasn't used in PR ;)

closed 2022-10-13 by pkendall64

MrD-RC, 2025-01-09:
> -1 is disabled.
> 
> That being said, the default should be -1 if the checkbox is disabled.

I know this is an old issue. But it would probably be worth including in the documentation that -1 is disabled and 0 is instant. It isn't mentioned and would be very helpful [here](https://www.expresslrs.org/quick-start/webui/).

naggie, 2025-05-25:
The web flasher unfortunately does not support `-1` -- it changes it back to zero.

pkendall64, 2025-05-25:
> The web flasher unfortunately does not support `-1` -- it changes it back to zero.

In the web-flasher you can empty the field and it say "Disabled" then it should work correctly.
Note that if you delete the data in the field, it may put a '0' there, just delete that then it should say "Disabled".

naggie, 2025-05-25:
Oh that's useful to know, thanks @pkendall64

PR #1798 (2022-08-30): Fix wifi auto on interval when not entered/selected in configurator
This PR should fix #1797 
Where the default was being set at 60 like we have in `user_defines.txt` as a default, but from configurator if not set then the firmware was using -1 (i.e. disabled).