ExpressLRS/ExpressLRS #1799: ExpressLRS Configurator V1.4.2 does not load Binding_Phrase onto receiver

RandMental, 2022-08-26:
## Current Behavior
ExpressLRS Configurator V1.4.2 does not load Binding_Phrase onto receiver

## Possible Solution (Not obligatory)
Load Binding_phrase using the Wifi website of teh RX device 

## Your Environment
ExpressLRS Configurator V1.4.2 

- TX hardware: Tx16s with Namimno 2400_ESP_OLED TX module 
- RX hardware: NamimnoRC_FLASH_2400_ESP_RX-3.0.0-RC2.bin
- Handset model:  TX16s running OpenTX 2.3.15
- OpenTX version (including nightly number)  OpenTX 2.3.15
- ExpressLRS version (TX & RX MUST MATCH): ESP_RX-3.0.0-RC2.bin
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2022-09-02:
How do you know it didn't "load binding phrase" onto receiver?
If you just looked at the web page on the RX after flashing, then you will not see your binding phrase, but the UID underneath will show the correct values for your binding phrase. i.e. the binding phrase is just a nice way for a user to get the 6 byte UID that is actually stored and used by ExpressLRS.

RandMental, 2022-09-04:
With V3.0.0 RC2 after flashing the receiver won't bind unless I add the binding phrase in the Webpage and restart/reboot.
Flashing back to V2.5.1 the receiver binds immediately.

I confirmed and tested this more than once, ensuring the correct and same Binding phrasese are used during the TX and RX flash process.

nightflasher, 2022-09-11:
Same here, with the RP1 (radiomaster) receiver. First time flashing the receiver (got it new on edgeTX Event on Saturday), the bidding phrase wasn't stored in the receiver. I know for sure the uid for my phrase and saw it immediately., but changing in web Ui worked fine.

pkendall64, 2022-10-30:
This has been fixed by #1911 and will be in release 3.1
If you'd like to test, just flash the master branch

closed 2022-11-13 by pkendall64