ExpressLRS/ExpressLRS #1801: V3 Lua does not update RSSI dBm sensitivity label until after it is selected

bede, 2022-08-28:
## Current Behavior

The 3.0.0 RC2 Lua script (that which is bundled with 3.0.0 RC2) does not show the correct RSSI dBm sensitivity threshold when selecting different packet rates and modulations *until after that mode is selected*. This feels like a regression from the V2 Lua behaviour (bundled with 2.5.1) where the threshold is updated when scrolling between modes as one might expect.

The sensitivity label update function is defined here in `tx_devLUA.cpp`:
https://github.com/ExpressLRS/ExpressLRS/blob/e7245ad05ede0c89f6517bef0b25ba4d99e2b7c7/src/lib/LUA/tx_devLUA.cpp#L252

I'd like to work towards a PR if possible, but thought I'd write this down for now.

## Your Environment

ES24TX-Lite nano module and power board

- TX hardware: HM ES24 TX-Lite
- RX hardware: HM EP1
- Handset model: Jumper T-Lite v1
- OpenTX version: 2.3.15
- ExpressLRS version (TX & RX MUST MATCH): Matching wither 2.5.1 or 3.0.0 RC2
- Packet Rate: Any
- Telemetry Ratio: -
- user_defines: -

0crap, 2022-08-31:
JB and I agree this was better before. Watch at 09:00 min.

https://youtu.be/cyILQqxPpZI?t=540

bede, 2022-08-31:
Ha, Bardwell has called it out, officially a bug 🤣

AlessandroAU, 2022-09-03:
Fixed

closed 2022-09-03 by AlessandroAU