ExpressLRS/ExpressLRS #1805: trying to flash TX16s with 3.0RC2 and it says "bad size given" while trying to update over WIFI.

gminton86, 2022-08-30:
I tried the 3.0 RC1 and RC2 and they both say the same thing 2.5.1 works fine.

pkendall64, 2022-08-31:
This seems to be a common issue and users who have this problem will have to perform an updated using USB.

pkendall64, 2022-09-02:
We've figured out the issue and provided a solution in the pinned comments in the #help-and-supprt channel on discord.

gminton86, 2022-09-02:
Thanks I got it to work with your advice.

jurgelenas, 2022-09-02:
> We've figured out the issue and provided a solution in the pinned comments in the #help-and-supprt channel on discord.

is it documented on expresslrs.org?

pkendall64, 2022-09-02:
> > We've figured out the issue and provided a solution in the pinned comments in the #help-and-supprt channel on discord.
> 
> is it documented on expresslrs.org?

No, not yet. Waiting to here back from deadbyte and others on their testing efforts.