ExpressLRS/ExpressLRS #1807: ELRS 3.0 RC2 Flash Fail with Radiomaster Zorro/ TX12. I reflashed ELRS 2.51 and the flash completed with Success as it's supposed to.

burgerplanet, 2022-08-31:
## Current Behavior

ELRS 3.0 RC2 Flash Fail with Radiomaster Zorro/ TX12. I reflashed ELRS 2.51 and the flash completed with Success as it's supposed to.
## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2022-08-31:
You have not provided any useful information in this bug report! I assume you got the dreaded "Bad size given" message.
If that is the case then you need to flash via USB and this is very easy to do with the Zorro, just follow the instructions here https://www.expresslrs.org/2.0/quick-start/transmitters/rm-internal/#flashing-via-edgetx-passthrough

pkendall64, 2022-08-31:
Also, you should as for help on discord about this in the #help-and-support channel.

burgerplanet, 2022-08-31:
> You have not provided any useful information in this bug report! I assume you got the dreaded "Bad size given" message. If that is the case then you need to flash via USB and this is very easy to do [...]

That's what I tried originally. It says I have EdgeTX 2.8.0, but only version 2.7.1 is available online? Im using the new Radiomaster TX12 MKII

pkendall64, 2022-08-31:
They have probably flashed a pre-release version onto the TX16S, so you should be ok to use that version and just follow the other instructions to flash the ELRS module.

closed 2022-09-02 by pkendall64