ExpressLRS/ExpressLRS #1812: VTX settings are changed mid-flight if VTX Administrator is enabled in the LUA script

NameOfTheDragon, 2022-09-04:
## Current Behavior

I have observed this issue on two different models both running Betaflight 4.3.1.

I set my VTX power using the 6-way switch on my radio. I will select the power level I want, then arm and take off. The model will fly for a variable length of time (sometimes a whole flight, sometimes 10 to 20 seconds) before the VTX power level spontaneously changes and/or pit mode is enabled. Sometimes this results in a complete loss of video and a crash, other times just a reduction in VTX power.

I realized what the issue was when I saw a video in which someone else had the same problem. He said to make sure the VTX Administrator was set to "off" and sure enough that fixed the issue on both models. Now, the VTX setting never changes unless I press a button on the radio. The video from CurryKitten that gave me my "Aha moment" is at https://youtu.be/KXSa92oJ8vs?t=779 - you can see that he sets the power level using the OSD menu, takes off, then the VTX changes mid-flight. This is a very clear illustration of the issue, although I have found that sometimes I can go a whole flight without it happening and then it will happen typically on the second flight. KurryKitten presents the results of his research near the end of the video and this led me to check my VTX Administrator settings and sure enough pit mode was set to ON and power level to 25mW, and these settings were somehow getting applied mid-flight exactly as shown in the video. I turned these settings off and the problem immediately went away on both models.

The ELRS documentation states: "VTX Administrator will only send data when disarmed". This is clearly not so, or if it is, then the receiver is acting on information that has not been semt. Either way, I believe this is a pretty serious issue that will result in crashes.

## Steps to Reproduce

1. Use the LUA script VTX Administrator to set the VTX channel and band, and set PIT mode to ON.
2. Select [Write VTX]
3. Exit from LUA script
4. Use the Betaflight OSD menu or radio controller to configure the VTX differently (i.e. turn off pit mode, set a higher power level)
5. Arm and take off.

Perhaps on the first flight, but more usually on the second, the VTX settings will change mid-flight while the model is armed.

## Video of the issue ##

This isn't my video but it is a very clear example of the problem happening. Please see https://youtu.be/KXSa92oJ8vs?t=779

## Possible Solution (Not obligatory)

The documentation states that ELRS will not change VTX settings while the model is armed. This is clearly not the case so somewhere there is a bug and ELRS is not behaving as documented. I believe this is a critical safety issue because it could cause the pilot to lose control and crash the aircraft.

## Your Environment

- TX hardware: HappyModel ES24TX-PRO
- RX hardware: HappyModel EP1 on two different models
- Handset model: Radiomaster TX-16S (V1) 
- OpenTX version (including nightly number) EdgeTX v2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.0 RC2
- Packet Rate: 150Hz
- Telemetry Ratio: STD
- user_defines:

JyeSmith, 2022-09-04:
Sounds like everything is actually working as expected.  The one thing to know is that the VTx Admin will send the vtx command on every tlm connection.  This should be when a new craft is powered and connects, or if you are using a high tlm ratio (1:128) you may have a tlm intermittently lost/recovered and the vtx command is resent.

ELRS and BF allow vtx power changes while armed for future features like dynamic vtx power.  But you are setting the vtx power via an aux.  So if you do not want the VTx Admin to be used at all set the freq selection to '-', and nothing will be sent.  If you only want band/channel to be sent then set power to '-'.

NameOfTheDragon, 2022-09-05:
> Sounds like everything is actually working as expected

Respectfully, I disagree. This does not resolve the problem. The documentation clearly states that 
[image]

I believe that is the correct and safe approach.

I agree and understand that Betaflight allows teh VTX settings to be changed in flight (indeed, that is what I want because I control it with my 6-way switch). However, once armed, it is highly dangerous to allow the VTX Administrator to make unexpected changes without user input.

I would request that you NOT allow this to happen while the model is armed.

JyeSmith, 2022-09-05:
This is a documentation issue and needs to be updated.

NameOfTheDragon, 2022-09-05:
I can't stress enough how much I think it is a mistake to allow VTX settings to be written based on telemetry loss. It results in highly unexpected nondeterministic behaviour.

bike797, 2022-09-15:
> ELRS 및 BF는 동적 vtx 전원과 같은 미래 기능을 위해 무장하는 동안 vtx 전원 변경을 허용합니다. 그러나 Aux를 통해 vtx 전원을 설정하고 있습니다. 따라서 VTx Admin을 전혀 사용하지 않으려면 주파수 선택을 '-'로 설정하면 아무 것도 전송되지 않습니다. 대역/채널만 전송하려면 전원을 '-'로 설정하십시오.

How about changing the following to solve the above problem?
  - disarms : vtx admin can change all parameters
  - arming: vtx admin can change only power level (cannot change channel or band)

pkendall64, 2023-04-01:
A clarification has been added to the documentation repository.

https://github.com/ExpressLRS/Docs/pull/304

closed 2023-04-01 by pkendall64