ExpressLRS/ExpressLRS #1814: Multiple failsafes per second during flight with 3.0 RC2 on 868Mhz

Auge103, 2022-09-04:
## Current Behavior

I upgraded to ELRS 3.0.0 RC2, coming from 2.5.x (also the script ofc), confirmed everything working, did a small range test outside and today proceeded to test fly the firmware.

During an entirely normal flight the copter failsafed behind a tree, got into GPS rescue RTH and continued to switch between signal and failsafe multiple times per second, even at height within VLOS, for the entire remaining duration of the flight, at every distance starting from 1km down to <10m to the pilot/TX.
I was barely able to get my copter back, since the fast switching between signal and failsafe disrupted not only the pilots input, but also the correct function of GPS RTH failsafe mode.
There was no mechanical damage, nor any loose antennas on the TX/RX side and a very quiet RF environment (I was >3km away from the nearest settlement)

## Steps to Reproduce

Sadly I have no idea how to reproduce this issue. 
I've flown 10+ batteries in the hours before this happened and immediatly after I landed, I switched to another battery and the copter flew as it should again, all issues disappeared completely after that powercycle.

## Details
Flight video from the time the issue began until disarm: https://www.youtube.com/watch?v=Aen0JAqRPi8
TX/handset telemetry log: https://pastebin.com/jkjAn1yV

Range between TX and RX at the point of failure was about ~1km, same height, RX behind a tree, TX on open field
No settlements or any technology in a radius of >3km
Flighttime before the issue occured was about 3 minutes
Copter flew without issues on this firmware and location for 10+ packs before the issue occured
Copter flow without issues on this firmware and location after the issue occured, with no changes to the hardware
TX baudrate is at 921khz
The TX module was set to 500mW at dynamic power
Ambient temperature was about 20°C and very windy
I wiggled on the TX antenna and turned myself around in a circle while the issue persisted, no change.
DJI goggles OSD did show RSSI/LQ warnings and constant connection loss from the start of this issue until disarm
TX handset put out voice warnings about LQ degraded and link lost multiple times.

## Your Environment

- TX hardware: BetaFPV ELRS Micro 868Mhz | BetaFPV 868Mhz Antenna
- RX hardware: BetaFPV ELRS Nano | TBS Crossfire Immortal T Antenna V2 | Mamba MK4 h743 flight controller
- Handset model: Radiomaster TX16s MKII
- EdgeTX version: edgetx-tx16s 2.7.1 (a55aff02)
- ExpressLRS version: 3.0.0-RC2
- Packet Rate: 200hz
- Telemetry Ratio: 1:8
- user_defines: REGULATORY_DOMAIN_EU_868 || BINDING_PHRASE || UART_INVERTED || UNLOCK_HIGHER_POWER || AUTO_WIFI_ON_INTERNAL

nightflasher, 2022-09-11:
Same here with EVERY rx (diy 20x20 868MHZ & frsky r9mx & r9mm/r9mini) and the ttgo_v2 as TX module.
https://youtu.be/zzZYxMYnAqU

adamiss5138, 2022-12-14:
Same here. R9M (2019) module and R9 mini receiver. Both ELRS version 3.0.1 (868 MHz EU LBT). Tx power set to 250mW and I failsafe behind a fcking tree and dropped 20m from the sky. I was only about 40 meters far, video was totally okay. I never thought I would failsafe ELRS 40 meters away from me.
Antenna is not broken, because I was running stock FrSky firmware before without any problem.
Now I got hard time trusting ELRS ever again.

AlessandroAU, 2022-12-29:
@Auge103 failsafe not unexpected as signal was dropping to -108dbm! just before link loss. Logs do not show TX power so unable to confirm dynamic power was working as expected. Possible antenna issue?

@nightflasher you need to use RSSIdbm to gauge signal strength, LQ does not tell the full story. You should have turned around when LQ dipped into the signal digits. LQ drops also coincident with when you do turns suggesting a combination of poor signal and cross-polarisation antenna loss. Again, possible antenna issue. 

@adamiss5138 Be aware the R9M module is prone to overheating, I can't say for sure that's what happened with you but all the close range R9M failures I've seen were due to overheat or damaged coax. Suggest to use 100mw and monitor signal with RSSIdbm to confirm good reception.

pkendall64, 2023-04-01:
Any more info on this issue or should we close it. There have been quite a few fixes to LBT since 3.0 and even up to the new 3.2.1 release about to come out.

closed 2023-04-14 by github-actions[bot]