ExpressLRS/ExpressLRS #1818: Can't exit failsafe on 3.0.0-RC2

SolkGit, 2022-09-05:
## Current Behavior

Can't exit failsafe after control link is regained on 3.0.0-RC2.

## Steps to Reproduce

1. Radio control on
2. Quad on
3. Arm motors -> motors spin
4. Radio control off -> 'fail safe' blinking in osd, motors off
5. Radio control on -> control link regained, 'fail safe' still blinking in osd
6. Arm motors -> motors off, 'failsafe' is shown (not blinking, without space)

## Possible Solution (Not obligatory)

Use earlier versions e.g. 2.5.0 untill the problem is solved.

## Details

Release 2.5.0 works fine, but 3.0.0-RC2 fails to exit failsafe mode. As regulatory domain I use EU_CE_2400. Maybe it is due to new LBT feature, but also using ISM_2400 failsafe mode can be exit in 1 of 10 tests.

## Your Environment

- TX hardware: BETAFPV 2400 TX Micro 1000mW
- RX hardware: HGLRC HERMES 2400 RX or BETAFPV Nano 2400 RX
- Handset model: Eachine TX16S
- OpenTX version (including nightly number): EdgeTX 2.6.0
- ExpressLRS version (TX & RX MUST MATCH): 2.5.0 and 3.0.0-RC2
- Packet Rate: 500 Hz
- Telemetry Ratio: 1:128
- user_defines: no

[image]
[image]
[image]
[image]

pkendall64, 2022-09-06:
You have not said which flight control firmware and version you are using.

SolkGit, 2022-09-06:
> You have not said which flight control firmware and version you are using. You also say set telemetry to off as a possible solution, and then say it does not recover if telemetry is set to off. Can you clarify which one is correct. Or is it that on EU_CE, telemetry off works and ISM2400 telemetry off does not recover?

Hello, thank you for the hint regarding flight control firmware. I'm using betaflight 4.2.0 and 4.3.0, added the information to the description. I said "CAN exit failsafe, if telemetry ratio is set to 'off' using 3.0.0-RC2"

SunjunKim, 2022-09-08:
Do you have any DVR demonstrating the problem?

SolkGit, 2022-09-08:
> Do you have any DVR demonstrating the problem?

I added some recordings in Details

SolkGit, 2022-09-08:
> Seems to all be working according to spec to me. Can you provide a test case that is not working to spec? And how does telemetry have anything to do with anything, are you running VTX Admin that might be mucking up the reconnect process maybe?

I've added some recording where you can see the difference between telemetry settings in 3.0.0-RC2 (Off vs Std)

CapnBry, 2022-09-08:
Wow, what the fuuuuuuuuudge is going on there? Everything works as I described and then it goes from FAILSAFE to FAIL SAFE. Iiiiii do not know what the heck is going on there. I'll have to check into it deeper tomorrow. Have you tried upgrading to Betaflight 4.3.1? All my testing was in 4.3.1 and it re-armed every time.

CapnBry, 2022-09-09:
Yeah I can't reproduce your behavior at all, using the exact same settings down to the LBT, ReceiverID and VTX Admin settings. It works exactly as expected, which is that it comes back with BADRX after failsafe, I turn off the arm switch, and it is ready to go again. My quad doesn't have a GPS though so that might change the behavior slightly? Doesn't look like there is one in the video either though.

Suggestions to try (in this order):
* Update the FC to Betaflight 4.3.1
* Setting VTX Admin in the Lua to Off (this relies on telemetry being on so there's a possible interaction here, but it worked with it enabled for me)
* Disable the GPS feature in the Configuration tab of Betaflight if enabled

SolkGit, 2022-09-09:
> Yeah I can't reproduce your behavior at all, using the exact same settings down to the LBT, ReceiverID and VTX Admin settings. It works exactly as expected, which is that it comes back with BADRX [...]

I'm testing different settings now and will provide the results as fast as possible. With ELRS 2.5.0 I also could exit failsafe mode if I keep the arm switch ON during power up of radio controller, but can't if the arm switch is already OFF.

SolkGit, 2022-09-09:
> * Update the FC to Betaflight 4.3.1
Tested with 4.3.1, same results as with 4.3.0. Noticed "BST" warning. Check the recording Update (09.09.2022 19:26).

> * Setting VTX Admin in the Lua to Off (this relies on telemetry being on so there's a possible interaction here, but it worked with it enabled for me)
How to disable VTX Admit in the Lua script. I see only "Band", "Channel", "Pwr Lvl" and "Pitmode"

> * Disable the GPS feature in the Configuration tab of Betaflight if enabled
GPS feature is disabled.

SolkGit, 2022-09-09:
> * Update the FC to Betaflight 4.3.1

Also tested with iNav 5.1. Still no control, if radio is powered on with disarmed motors.

CapnBry, 2022-09-09:
> How to disable VTX Admit in the Lua script. I see only "Band", "Channel", "Pwr Lvl" and "Pitmode"

You set the Band to "Off"

I just can't think of how this could actually happen. Maybe you can run the same test you've been doing, then when it is stuck no able to get out of Failsafe, plug in a USB cable to the flight controller (without unplugging the battery) and see what it says the `Arming disable flags` is when you type `status` in the CLI tab.

SolkGit, 2022-09-09:
> I just can't think of how this could actually happen. Maybe you can run the same test you've been doing, then when it is stuck no able to get out of Failsafe, plug in a USB cable to the flight [...]

Motors disarmed: Arming disable flags: FAILSAFE CLI MSP
Motors armed: Arming disable flags: FAILSAFE CLI MSP ARMSWITCH

SolkGit, 2022-09-09:
> You set the Band to "Off"

After setting Band to "Off" I can regain control powering on the radio controller with armed and disarmed motors. Tested with ELRS 2.5.0

SolkGit, 2022-09-09:
> > You set the Band to "Off"
> 
> After setting Band to "Off" I can regain control powering on the radio controller with armed and disarmed motors. Tested with ELRS 2.5.0

Meanwhile official rELRS release 3.0.0 is available, so I tested it. With active VTX administrator control CAN'T be regained in both cases, power on with arm switch on and off. Switching VTX administrator off regaining control works in every situation.

dentex, 2022-09-13:
> Meanwhile official ELRS release 3.0.0 is available, so I tested it. With active VTX administrator control CAN'T be regained in both cases, power on with arm switch on and off. Switching VTX administrator off regaining control works in every situation.
Sorry about not adding much to the discussion, I was just scrolling through issues before upgrading to ELRS 3.0.0, because I have a working setup (with GPS) on 2.5.0... 
So can you confirm, just leave the VTX administrator off and we're good to go?

CapnBry, 2022-09-13:
> Meanwhile official ELRS release 3.0.0 is available, so I tested it. With active VTX administrator control CAN'T be regained in both cases, power on with arm switch on and off. Switching VTX administrator off regaining control works in every situation.

Thanks for doing all the testing and reporting back. Ok so we now know _where_ the problem lies but I'm not sure why it is happening still, or what the difference is between 3.x and 2.x that brings it up (both send VTX Admin info in the same way).

I'll have to dig into this some more and see if I can reproduce it somehow, as it still works for me even with VTX Admin on.

EhAye, 2022-11-01:
Disabling VTX Administrator as mentioned in this thread also seems to have fixed most of my problems.

For me, while inav is disarmed, if my yaw stick (channel 4) crosses over 1500 then it causes hickups in channel response, and eventually a failsafe within a couple seconds. The failsafe won't clear until i reboot the wing. 

- If i set EdgeTX's Yaw/Channel 4 "PPM Center" to 1600, THEN power up my wing, i'm able to move yaw around on the right side, but if i move it to the left it failsafes.
- If i set EdgeTX's Yaw/Channel 4 "PPM Center" to 1400, THEN power up my wing, i'm able to move yaw around on the left side, but if i move it to the right it failsafes.
(Moving my stick past 1500, to the opposite 'side' will trigger the failsafe)

The initial symptoms were random failsafes, and i could trigger the failsafes with some stick commands. I think the random failsafes were caused by Yaw stick jitter pushing it past the 1500 mark, triggering a failsafe. And SOME of my "stick commands failsafes" were simply me moving the stick far to the corner, past 1500.

I think other people reporting random failsafes are probably also having the yaw jitter push it past 1500.

This only happens while disarmed. Once i've armed, the "yaw 1500 failsafe" doesn't happen and everything seems to function normally.
Disabling inav's telemetry also seems to fix the "yaw 1500 failsafe".
**But if i disable VTX Administrator, i can leave inav telemetry enabled.**

I'm still having other failsafe issues but i think those are due to a different problem with the h743-wing and inav memory writes, as they are only triggered by very specific stick commands.

I'm using a Matek h743-Wing V2, inav 5.1, ELRS 3.0.1, HM EP1.
Also tried on Matek D24D, inav 4.1, ELRS 2.5.1. Same issues.

CapnBry, 2022-11-02:
That is certainly a lot of good information and troubleshooting to narrow it down!

I run VTX Admin though and also have a Yaw stick that I use with no issues, so it doesn't seem intrinsic to ExpressLRS itself. VTX Admin does send commands on every RX connection though:
* Set VTX channel / band and optionally power / pit mode
* "Save EEPROM" (MSP command)

Is it possible that the yaw channel causes a disconnect or failsafe in iNav and then it can not clear that until after it can do an EEPROM save or something? It seems to me that there's something else in play here that's keying off your stick position, especially since everybody uses the yaw channel. I do not believe _other people reporting random failsafes are probably also having the yaw jitter push it past 1500._, I think this is very specific to your configuration.

Have you tried saving a backup of the iNav config, doing a fresh install with chip erase, just enable the Serial RX, then creating a new model on the handset that just sends CH1-CH5 with iNav telemetry, see if it fails? Then add the VTX setup to iNav, test it again. See if VTX Admin works properly. Then keep adding your saved config backup until it starts failing when you move the yaw stick.

pkendall64, 2022-11-02:
Quite possible related to, or caused by #1773

closed 2023-04-01 by pkendall64