ExpressLRS/ExpressLRS #1824: v3.0.0 bad size for HM ES24TX Pro

wernight, 2022-09-09:
## Current Behavior

Flashing v3.0.0 to my HM ES24TX Pro Series 2400 TX gives:

```
TARGET MISMATCH
Current target: HAPPYMODEL_ES24TX_PRO_SERIES_2400_TX.

Flashing the wrong firmware may lock or damage your device.
WIFI upload FAILED!
*** [upload] Error -2
========================= [FAILED] Took 156.15 seconds =========================

Environment                                    Status    Duration
---------------------------------------------  --------  ------------
HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI  FAILED    00:02:36.154
```

Flashing via the web UI gives a "Bad Size".

Note: I could flash again v2.7.1 without issue.

## Steps to Reproduce

1. Enable WiFi via LUA
2. Use ELRS Configurator to flash

## Possible Solution (Not obligatory)

The hash might be incorrect.

## Details

It's a HM ES24TX Pro, and I could flash the older release again without issue. So I don't think I made a mistake in selecting the wrong target for example.

## Your Environment

Running ELRS Configuration v1.4.2.

- TX hardware:  HM ES24TX Pro
- Handset model: TX16s
- EdgeTX 1.7.1

jesuslg123, 2022-09-09:
Have you follow the release notes? 

https://github.com/pkendall64/repartitioner/

wernight, 2022-09-09:
Oh thanks. I was relying on my notes to upgrade and the configurator doesn't show these I think

mgrennan, 2022-09-09:
This is an issue for the BetaFPB ELRS 2.4GHz 1W Transmiter too.

I can't figure out how to build this binary even with Local repository tab.   Can you build it for us?

pkendall64, 2022-09-10:
The repartioner binary is in the release on github. All built and ready to go.

Shouldercannon, 2022-09-11:
> Oh thanks. I was relying on my notes to upgrade and the configurator doesn't show these I think

"Bad Size" did you see in the browser itself?
The firmware loads suspiciously fast.

closed 2022-09-14 by pkendall64