ExpressLRS/ExpressLRS #1834: "Telemetry lost/telemetry recovered" loop with ELRS 3.0.0/2.4g LBT/LoRa/EdgeTX 2.8 nightly

froqstar, 2022-09-12:
## Current Behavior

At some point (can be directly after turning equipment on, or after some minutes of flying), I get callouts for telemetry being lost, followed by telemetry being recovered, then lost again and so on, with what feels like increasing frequency, ending up with lost/recovered loops happening nearly every second.

Along with the callout by EdgeTX, the `C` in the ELRS Lua disappears and reappears, and the telemetry in the model telemetry page freezes and turns red while the connection is lost.
Control input still works fine and the model is fully controllable, minus telemetry of course.

The issue only seems to appear with LoRa modes 150Hz and up.

## Steps to Reproduce

- turn on TX, turn on RX, wait some minutes (usually <<5min for me)

## Possible Solution (Not obligatory)

- switch to a FLRC mode
- switch to a fullres LoRa mode
- switch to 50Hz LoRa mode

Those modes seem to be unaffected, but 150Hz, 250Hz and 500Hz LoRa modes experience this problem.

## Details

[EdgeTX logs](https://github.com/ExpressLRS/ExpressLRS/files/9551808/log-2022-09-12-224452.csv)
[image]
[image]
[video](https://user-images.githubusercontent.com/1944754/189763956-29c836bf-1356-478f-b05e-dad81444b2bb.webm)

## Your Environment

- TX hardware: TX16S MKII internal module (also reproduced with Happymodel ES24TX non-pro in a QX7 in the RC phase)
- RX hardware: Vantac Lite RX & Betafpv lite RX so far, multiple models of each
- Handset model: TX16S MKII, 1.87M baud
- OpenTX version (including nightly number): EdgeTX 2.8.0-nightly (1c715626) from 2022-09-11 
- ExpressLRS version (TX & RX MUST MATCH): 3.0.0
- Packet Rate: 150Hz (also happens on 250Hz & 500Hz LoRa modes)
- Telemetry Ratio: Std (doesn't seem to matter)
- user_defines: - all defaults except bind phrase & wifi credentials

Happens both on the bench at home (only 3 wifi networks around) as well as on my flying field.
Happens both with standalone RX as well as with INAV FC or Matek PWM boards attached.

CptLuki, 2022-09-13:
Just flashed 3.0 and having similiar issue.
- Matek R24S RX
- Ghost Lite TX + Tango 2
- 250hz Lora
- 250mw
- 2.4g ism
- hybrid switches
- 1:64 tlm
Getting tlm lost tlm recovered warnings. It starts after arm and couple of seconds into flight. Pretty much doing circles around myself and getting these warnings. Tried 1:32 ratio and same thing. Tried to lower tlm power from 100mw to 50mw. Same thing. All was fine on 2.5.1.

Demon8Core, 2022-09-15:
I got the same problem with my Zorro, but I also lose the connection and it fail-saves like every second. 
Setup:
25mw
5.75M with 0 Err

BetaFPV micro PWM
d250hz
1:32 tlm

Radiomaster EP2 
d250hz
1:64 tlm

I also tried 50hz, but with no luck.

both receivers got the same problem, losing every second the connection...

Sekilsgs2, 2022-09-17:
Im have same issue with r9m

mattiamari, 2022-09-17:
I have the same issue using:
- Radiomaster zorro, edgetx 2.7.0, internal elrs
- elrs rx: happymodel EP2 with builtin ceramic antenna
- elrs v3.0.0, 500hz (did not test with other rates)

CptLuki, 2022-09-17:
Did some back and forth tests with 2.5.1 and 3.0.0. 
Can confirm that on 2.5.1 everything works as expected. Could not get tlm lost when walked to different rooms. Tlm LQ solid 80-90 through couple of walls. In flight could not get tlm lost in normal proximity flying up to at least 300m. Overall good LQ droping max to 85 with ceramic antenna. 

On 3.0.0 always getting tlm lost when I walk to next room. So basically just 1 wall causing tlm lost. In flight same thing. Couple of meters away getting into tlm lost tlm recovered loop when flying pretty close. When flying further like 200m I get tlm lost and it is not even getting back. Only when I come closer again. And overall LQ also seems to be lower. Was hitting LQ 40s while on 2.5.1 could not get under 85. So seems hardware is fine. Not sure what is wrong on FW side. 

- Matek R24S RX (ceramic antenna)
- Ghost Lite TX + Tango 2 ( latest fw 1.34)
- LoRa 250hz + 250 mW power

Not sure if that oscillator has something to do with it as Matek was in danger zone. Also that RX is using PA. And in 2.5.1 it was only 1 target for both Matek RXs. Now in 3.0.0 there are 2 targets. 1 specially for diversity RX R24D

robertgodlewski, 2022-09-17:
I have experienced same issue on 3.0.0 and LoRa 150Hz mode - constant "telemetry lost/recovered" and poor LQ despite good rssi (~-40dbm). No failsafe though. No problem after switching to FLRC 500Hz.
My setup:
- Taranis X9D with EdgeTx 2.7.1
- DIY 2.4GHz Tx
- Matek R24D
- CE LBT @ 50mW

pkendall64, 2022-09-18:
Would any of you be willing to test PR #1845 to see if makes a difference.

froqstar, 2022-09-18:
@pkendall64 I just flashed my TX16S MKII and a betafpv lite RX to this PR version, and it doesn't improve the behavior as far as I can tell. Thanks for the suggestion though.

0crap, 2022-09-18:
@froqstar Can you delete all sensors on the radio model in question and rescan for sensors?
Just to make sure your radio is not waiting for sensors that do not exist any more for some funny reason.
It's really weird you have no telemetry at all on 150Hz...

froqstar, 2022-09-18:
@0crap been there, done that, no change. but thanks for taking a look at this :)

In the meantime I think I found a clue @pkendall64 :
I flashed both TX and RX to ISM instead of EU_LBT domain and ran it for 20min at 250Hz LoRa, rock solid telemetry :).
Same for 500Hz LoRa so far.

I am just wondering why it affects some RF modes and others not at all? Or I just didn't hit that bug there _yet_...

0crap, 2022-09-18:
While I'm in the EU (NL) I never used EU_LBT, just sounds like a bad idea.
But OK, for the sake of testing I flashed my 2.4 GHz DIY module to 3.0.0 release EU_LBT.
(Radio is a Taranis QX7S ACCST with the hardware inverter mod.)
Also flashed a HM EP1 RX to 3.0.0 release EU_LBT.

Unfortunately I can't reproduce the issue.
On 150Hz LoRa I can go as low as 1:64 telem (46bps), all fine. (iNav 5.0 reporting 23 sensors total.)
Also tested LoRa 250 and 500Hz at 1:64, all fine.
On LoRa 50Hz I need to bump the telem ratio a bit because of the ridicules slow updates at 1:64.

froqstar, 2022-09-18:
I seem to have fixed it!? by playing with the dynamic power settings (turning it off/on, changing power levels). I think it started to disappear once I disabled dynamic power, but not entirely sure anymore, could just have been timing. 

Could be that I got corrupted data in those settings that needed to be overwritten to make it work? 
Is there a way to completely clear the TX module memory (apart from the firmware itself) to have a clean slate instead of trying to fix it by fiddling with those settings for all my models?

pkendall64, 2022-09-18:
Not sure if it could happen from corrupt data in the config, but yes, there is a way to reset the config i.e. all models and other settings available in Lua, back to defaults. Start wifi and visit `http://elrs_tx.local/reset?model` note that the hostname part may be different depending on if you network supports MDNS or if you are using access point mode.

0crap, 2022-09-19:
@froqstar Please let us know if the `reset?model` fixes your issues.

froqstar, 2022-09-19:
@0crap 
first an update _without_ `reset?model`: This morning, the issue I thought fixed yesterday just reappeared. 
Then I tried the suggested `reset?model`, and so far the problem didn't manifest in about an hour of testing on the simulator, with different LoRa rates. 
I also _think_ the controls are much smoother now, but that could be placebo.

froqstar, 2022-09-19:
And here we go again...the `reset?model` "fix" only worked for two hours or so, then the problem appeared again.
Next I tried the same (`reset?model`) with the RX, which again "fixed" it for about two hours give or take, and now it happens again...
Not so sure about the ISM domain firmware now, I may need to use that for an extended amount of time to be sure.

0crap, 2022-09-19:
Try the PR above as asked by pkendall64.
Take your time to see if that makes an improvement.

froqstar, 2022-09-19:
> Try the PR above as asked by pkendall64.
> Take your time to see if that makes an improvement.

Already tried that (https://github.com/ExpressLRS/ExpressLRS/issues/1834#issuecomment-1250232098) and it showed no difference in behavior, got telemetry loops almost instantly. 
I may just re-test that PR though because I have no other ideas anyway at the moment, maybe I did something wrong yesterday while testing it :)

froqstar, 2022-09-20:
Just a confirmation: 
ISM firmware version is stable for multiple hours of simulator flying.
Can you confirm my findings @robertgodlewski ?

robertgodlewski, 2022-09-20:
I've flashed ISM version today and do some tests on the desk - no issues at all. Next, I have flashed CE and after a while problem returned. For me only 150Hz LoRa mode is affected and tx power level (static and dynamic) does not make a difference. I've switched again to ISM and play with it a little bit longer - it worked flawless. I will do some flights over the weekends with ISM and share the results.

robertgodlewski, 2022-09-24:
Ok, over half an hour in air with ISM 150Hz LoRa mode and not a single issue/hiccup - solid link all the time. Power was set to dynamic but stayed at 10mW all the time. I'll stick to ISM for now.

pkendall64, 2022-09-24:
> Ok, over half an hour in air with ISM 150Hz LoRa mode and not a single issue/hiccup - solid link all the time. Power was set to dynamic but stayed at 10mW all the time. I'll stick to ISM for now.

Thanks for the report, that will help immensely in finding the problem!

zenobik, 2022-09-25:
I have the same issue with telemetry lost/recovered in an endless loop:

TX: Radiomaster Zorro EdgeTX 2.7.1, internal elrs
RX: Radiomaster EP1
ELRS v3.0.0, 150 Hz (1:32, 1:16), 250Hz (1:32, 1:64)
TX Power 100mW Dynamic EU_CE_LBT

I had no problems with the previously released version v2.5.1.
The overall range and link stability is much better in 3.0.0, but this telemetry issue is really annoying.

Demon8Core, 2022-09-25:
Hey Zenobik,
Can you try the ISM domain?
It also worked for me.

pkendall64, 2022-09-25:
Could you all that are having this problem please re-test pr #1845 as @JyeSmith and I have made some significant improvements to the timing.

alphabravogroup, 2022-09-25:
Same issue here, but reflashing both TX16s and Matek R24D seems to have fixed it.
Initially it would happen within 10 seconds of powering up changing any/all settings didn't seem to affect it.

robertgodlewski, 2022-09-25:
> Could you all that are having this problem please re-test pr #1845 as @JyeSmith and I have made some significant improvements to the timing.

Quick test on desk - no problems for me. More testing later.

froqstar, 2022-09-29:
Same here, finally came around to testing with the linked PR, and while it *may* have improved (hard to tell), I still get the telemetry loops after a few minutes.

pkendall64, 2022-10-05:
> Maybe @pkendall64 is onto something with #1867 ? After flashing commit `03d9d5` and a good hour of testing the 250Hz LoRa mode seems to be stable for me. 500Hz only hit "the loop" after 10 minutes or so. 150Hz is still broken, goes into looping mode almost immediately. I only flashed the TX, RX is still on 3.0.0 as that PR should not make a difference there?

You shouldn't need to update the RX with that PR.

Can you tell me all your settings on the module because I cannot, for the life of me, replicate the problem described in this ticket. I've just been fixing other issues that I see.

I really need these things, packet rate, telem ratio, switch mode, model match?, power settings (dynamic?) etc. for the mode that exhibits the worst behaviour for you. Also, what values do you see in the telem screen in the radio for 1RSS, RQLY, TRSS and TQLY.

froqstar, 2022-10-07:
Settings for worst behaviour:

Rate: 150Hz
Telem Ratio: Std (doesn't matter AFAIK)
switch mode: hybrid (doesn't matter as well)
model match: Yes
dynamic power: Yes, 100mW limit
EdgeTX latest nightly (I update every week or so, with no change regarding this bug)
Baud: 1.87M

Telemetry values, RSSI and Qly are averages but don't change very much:
1RSS: -45dB
2RSS: 0dB
RQly: 98%
RSNR: 14dB
RFMD: 5
TPWR: 25mW-100mW
TRSS: -48dB
TQly: 98%
TSNR: 14dB

Thanks a lot for your efforts!

pkendall64, 2022-10-08:
Thanks for all the details. I've gone through my collection of receivers and tested the BETAFPV lite and the Vantac and I have found that although the BETAFPV worked perfectly fine for me on you setting, the Vantac is not giving any telemetry back at all!
So that gives me something to look for, and hopefully this will clue us in to the problem.

pkendall64, 2022-10-08:
I think I've found the problem!
The RX is using LBT as well, which means that if it thinks the channel is in use it will skip sending telemetry. The problem is worse on 150Hz because that particular mode has the longest on-air-time when transmitting and I think the RX is seeing RSSI on the channel and is skipping the telemetry send!

I noticed this because I initially had the RX compiled as ISM rather than LBT and there was no problem. Once I compiled the RX with LBT the problem raised its head instantly.

froqstar, 2022-10-08:
The LQ displayed in the OSD only dips down to 97 or so when this problem occurs, so the TX seems to be unaffected by the problem?

Doesn't the TX/RX also only check the RSSI *before* transmitting, so the actual transmission time shouldn't make a difference since it comes after checking?
To me this sounds like the RX is picking the "wrong" point in time to check whether the channel is busy? 

If the problem is indeed on the RX, then your EdgeTX sync PR shouldn't have made a difference and it was just placebo on my part? I also did a testflight right now on 250Hz and I had the problem again, with the PR version on the TX. So just placebo/luck.

In the meantime, if I want to be EU compliant, a possible workaround would be to only use LBT firmware on the TX (as RQly seems unaffected) and ISM on the RX with the telemetry power limited to 10mW?

pkendall64, 2022-10-08:
I did a lot more reading and testing last night and there are a couple of bugs in the LBT code.
1. When selecting 10mW the RX still does LBT. If the RX only has 10mW then it does not do LBT.
2. There is no compensation for the LNA on amplified RXes, so the level clear-channel code is too sensitive i.e. it's checking for an RSSI 12dB higher than it should!
I've got fixes for these bugs, but it still does not fix the issue with 150Hz. And I know it's timing based because if I change the LoRa coding rate from `LI 4/7` to `LI 4/6` which gives us a shorter on-air-time, the 150Hz mode is solid!

pkendall64, 2022-10-08:
Can I get some testing and feedback on PR #1870 please. Hopefully this fixes the issues you guys are seeing with LBT mode.

froqstar, 2022-10-09:
As commented in the PR, the fix looks like it is a fix indeed :) 
Simulator testing will hopefully follow in the evening.

Again thanks so much for looking into this!

pkendall64, 2022-10-09:
> As commented in the PR, the fix looks like it is a fix indeed :) Simulator testing will hopefully follow in the evening.
> 
> Again thanks so much for looking into this!

You're more than welcome. Thanks for testing and creating such a good bug-report that made it reproducible.

nightflasher, 2022-10-09:
I had the same issues before, I've just changed the tlm_report_interval_ms to 200LU and everything is fine. 
[image]

froqstar, 2022-10-10:
@nightflasher your problem is most likely a separate problem that also affects ISM firmware and will be fixed with EdgeTX 2.8.

Belly3D, 2022-10-10:
I am also having this issue.
Zorro Internal ELRS 3.0 ISM
100mW static
150Hz
Std Telem Ratio (tried higher ratios to no effect)
Hybrid switch
Default TLM Report Interval
BetaFPV nano receiver 3.0 ISM (100mW telemetry)

Getting the telemetry lost/recovered loop after a couple mins into a flight with good RSSI even when flying only 20m away.

pkendall64, 2022-10-10:
There is a problem with 150Hz and LBT, in that there is not enough time window given the OTA parameters used for 150Hz. So the best option is to use 250Hz or the new 100Hz Full mode if you really need the range.

CptLuki, 2022-10-10:
Interesting. I have exactly same behaviour as Belly3D and it is also not LBT. 

Tango 2 + Ghost Lite ELRS 3.0 ISM
250mW static
250Hz
Std Telem Ratio (tried higher ratios to no effect)
Hybrid switch
Default TLM Report Interval
Matek R24S 3.0 ISM (100mW telemetry)

Exactly same that once in the air getting this tlm lost/recovered loop like 10m away. Basically doing circles around myself. 

I noticed that both of us use receiver with LNA+PA. I wonder if this has something to do with it. There is no issue in 2.5.1. And again this is in ISM domain.

pkendall64, 2022-10-12:
More changes have been added to the PR and it would be good if we could get some testing and feedback.

froqstar, 2022-10-13:
@pkendall64 from your comments on the PR I understand that all RF modes (including FLRC) need re-testing, correct? 
Does it make sense to test 150Hz as well or is that a lost cause?

pkendall64, 2022-10-13:
> @pkendall64 from your comments on the PR I understand that all RF modes (including FLRC) need re-testing, correct? Does it make sense to test 150Hz as well or is that a lost cause?

Yes, all modes should work now, and should be tested.

Sekilsgs2, 2022-10-13:
But why all prs for 2.4 - im have troubles with 900 too - some times many telemetry lost , with 2.5.1 im never seen this. Any comments?

pkendall64, 2022-10-13:
> But why all prs for 2.4 - im have troubles with 900 too - some times many telemetry lost , with 2.5.1 im never seen this. Any comments?

Because this is a ticket about LBT which is only supported on 2.4 currently.
Some of the changes in the PR will also help 900Mhz, but that has not been my focus as I've been trying to fix the problems reported in the ticket, which is LBT issues.

closed 2022-10-13 by JyeSmith

Sekilsgs2, 2022-10-13:
> Because this is a ticket about LBT which is only supported on 2.4 currently. Some of the changes in the PR will also help 900Mhz, but that has not been my focus as I've been trying to fix the [...]

Have one separate issue for 868 devices - please read and comment - why 2x have stable link but with 3.x we have many troubles - maybe problems in timings - changed in 3x ?

Berlinologe86, 2022-10-15:
i have the same problem and a friend too 1.setup: frsky xlite+happymodel ex24pro on edgetx2.7 rx is a betafpv clone(150,250hz)tlm std.output power dynamic and static 500/1000mw tested 2. setup jumper tpro on edgetx2  .7 rx are original betafpv nano/lite modules(150-250hz) output power static 500/1000mw ans dynamic. Team 2.4

pkendall64, 2022-10-15:
> i have the same problem and a friend too 1.setup: frsky xlite+happymodel ex24pro on edgetx2.7 rx is a betafpv clone(150,250hz)tlm std.output power dynamic and static 500/1000mw tested 2. setup [...]

Ok, so can you test the PR #1870 to see if it fixes the problem for you and report on the PR with your findings please.

Berlinologe86, 2022-10-18:
But we have flashed ism

froqstar, 2022-10-18:
@Berlinologe86 can you try updating to EdgeTX 2.8 and test again? It includes a fix for handset/module communication.
Alternatively you can try setting the telemetry report interval in the module ELRS  firmware options to 200, which should also improve telemetry reliability.

BorisDess, 2022-10-19:
Issue encountered as well with TX16S MKII and GEPRC nano SE, both flashed with ELRS 3.0.

Berlinologe86, 2022-10-22:
I have test ist today edgetx2.8 rc3 with The Same settings. telemetry lost is still there.  I noticed that despite a dbm of -64, my jumper then has to boost to 500mw at a distance of less than 200 meters🙄Von meinem iPhone gesendet

pkendall64, 2022-10-22:
This has not been released yet! Did you flash the PR for testing?

atagunov, 2022-11-06:
Not much knowledge/time to test custom builds but I'm seeing the issue with 3.0.1 ExpressLrs on BetaFpv LightRadio 3 Pro with ExpressLrs 2.4 GHz internal module + Matek 24 RX I'm having a telemery lost loop with the quad just sitting on my table a few centemeters from the RX. REGULATORY_DOMAIN_EU_CE_2400, TLM_REPORT_INTERVAL set to 100LU.

nigerninja, 2022-11-09:
Hi, 
Thanks to the ELRS team looking at this.
I also have the same issue with the telemetry lost/recovered endless loop:

TX: Radiomaster Zorro EdgeTX 2.7.0, internal elrs
RX: BetaFPV ELRS Lite
ELRS v3.0.1, 150 Hz (1:32, 1:16), 250Hz (1:128)
TX Power 100mW Dynamic Off EU_CE_LBT

For what it's worth, just got a quad and spent the whole day setting up and getting familiar with BF and didn't start getting that loop until how at least 3 - 4 hours after. Initially thought I messed up something in BF but saw similar reports online. I'll try an ISM build in a bit.

Thanks.

nigerninja, 2022-11-10:
A quick update, ISM works perfectly no telemetry losses in the last hour or so.

Thanks.

tfranzel, 2022-11-19:
Can confirm that updating from **3.0.1** to **3.1** fixed the issue. no more "Telemetry lost/telemetry recovered". thanks a lot :heart: 

TX: Radiomaster Zorro EdgeTX 2.7.1, internal elrs
RX: Radiomaster EP1 
ELRS v3.1 regulatory domain EU CE

Dosclic98, 2022-12-01:
Having the same issue on the ISM regulatory domain version, but seems less present and with telemetry at 1:4, 500Hz and 250mW it happens just at range, while on version 2.5 I never had any issue.

OS: EdgeTX 2.8.0
TX: BetaFPV 2.4GHz Micro TX module 
RX: Happymodel EP1
ELRS v3.0.1 regulatory domain ISM

[16 more replies]