ExpressLRS/ExpressLRS #1843: ELRSv3 script exit with error when selecting ArduPilot devices

urlu75, 2022-09-16:
## Current Behavior

with ELRSv3.lua, when selecting other device and then ArduPilot device, always get this error
Script syntax error ?:0 attempt to index field '?' (a nil value)

Getting the ArduPilot parameter works well with ELRSv2.lua script.

## Steps to Reproduce

1. run ELRsv3.lua script when tx and rx are on
2. select other devices
3. select ArduPilot device (copter or plane)
4.error

## Possible Solution (Not obligatory)
works well with v2 script.

## Details

[image]

[image]

## Your Environment

- TX hardware: HM 2.4 & Namimno rc 2.4
- RX hardware: JHEMCU RX24T & Siyi FR mini
- Handset model:X9D & TX16s
- OpenTX version (including nightly number) 2.3.15 on X9D & Edge 2.6 on TX16s
- ExpressLRS version (TX & RX MUST MATCH): 3.0
- Packet Rate: 250hz
- Telemetry Ratio: 1:4
- user_defines: default

devimauz, 2022-10-29:
The problem at line 489: `if functions[field.type+1].load then`

wsnfly, 2022-12-07:
我有同样的问题
[image]

wsnfly, 2022-12-08:
> 第 489 行的问题：`if functions[field.type+1].load then`

How should it be modified? i have the same problem
`
local function parseParameterInfoMessage(data)
  local fieldId = (fieldPopup and fieldPopup.id) or loadQ[#loadQ]
  if data[2] ~= deviceId or data[3] ~= fieldId then
    fieldData = {}
    fieldChunk = 0
    return
  end
  local field = fields[fieldId]
  local chunksRemain = data[4]
  -- If no field or the chunksremain changed when we have data, don't continue
  if not field or (chunksRemain ~= expectChunksRemain and #fieldData ~= 0) then
    return
  end
  expectChunksRemain = chunksRemain - 1
  for i=5, #data do
    fieldData[#fieldData + 1] = data[i]
  end
  if chunksRemain > 0 then
    fieldChunk = fieldChunk + 1
  else
    loadQ[#loadQ] = nil
    -- Populate field from fieldData
    if #fieldData > 3 then
      local offset
      field.id = fieldId
      field.parent = (fieldData[1] ~= 0) and fieldData[1] or nil
      field.type = bit32.band(fieldData[2], 0x7f)
      field.hidden = bit32.btest(fieldData[2], 0x80) or nil
      field.name, offset = fieldGetString(fieldData, 3, field.name)
      if functions[field.type+1].load then                                    //here
        functions[field.type+1].load(field, fieldData, offset)
      end
      if field.min == 0 then field.min = nil end
      if field.max == 0 then field.max = nil end
    end`

pkendall64, 2022-12-08:
This may have been cause by the bug that we just fixed in the 3.x.x maintenance branch. Could you flash and test that for us?

wsnfly, 2022-12-08:
> 这可能是由于我们刚在 3.xx 维护分支持中修复的错误引发的。你能为我们闪现并测试一下吗？

https://www.youtube.com/watch?v=JCz5NcxoV2A&list=PLnSoxLtSdtqHxD8erohmGywcj6KBwsgn9&index=3
这个视频里复现了该错误。但我没有能力将这个BUG解决掉

pkendall64, 2022-12-08:
> > 这可能是由于我们刚在 3.xx 维护分支持中修复的错误引发的。你能为我们闪现并测试一下吗？
> 
> https://www.youtube.com/watch?v=JCz5NcxoV2A&list=PLnSoxLtSdtqHxD8erohmGywcj6KBwsgn9&index=3 这个视频里复现了该错误。但我没有能力将这个BUG解决掉

I see from your video that you have 3.0.0 firmware. Can you please try with 3.x.x-maintenance branch.

closed 2023-04-14 by github-actions[bot]