ExpressLRS/ExpressLRS #1848: Only 400k baud on 3.0?

lesharodin, 2022-09-18:
## Current Behavior

After flashing 3.0 on my HM ES24TX PRO it only works on 400k baud rate.
Flashed back to 2.5.1 all bauds works even without rebooting TX.
Flashed 3.0 - only 400k, 3.x.x maintanance - only 400k.

## Steps to Reproduce

Flash 3.0, change baud rate.

## Possible Solution (Not obligatory)
I think smth about that not working correctly
https://github.com/ExpressLRS/ExpressLRS/commit/f3773a74e6c275b1e3c306eb997fa33a1bab55eb

## Details

I tried both onebit and normal sample mods, doesnt change anything.

## Your Environment

- TX hardware: HappyModel ES24TX PRO
- RX hardware: -
- Handset model: Radiomaster TX12
- OpenTX version (including nightly number) 2.7.1 EdgeTx
- ExpressLRS version (TX & RX MUST MATCH): 3.0
- Packet Rate: 500
- Telemetry Ratio: std
- user_defines: -

pkendall64, 2022-10-06:
I have the same devices here and hand tested with the model CRSF set to 3.75Mbaud and it works fine.
I do remember at one stage that you needed to reboot the radio when changing the baud rate in EdgeTX.

lesharodin, 2022-10-08:
> I have the same devices here and have tested with the model CRSF set to 3.75Mbaud and it works fine. I do remember at one stage that you needed to reboot the radio when changing the baud rate in EdgeTX.

On 2.5.1 It works without rebooting, on 3.0 after I change from 400k to any baud LED on TX just rainbows all time, and lua not working

pkendall64, 2022-10-08:
If the RGB is not going to a single color and "rainbowing" all the time then there must be a boot loop.
Could you try flashing with the web-flasher at https://pkendall64.github.io/elrs-web-flasher/ and select the "full erase" option when flashing.