ExpressLRS/ExpressLRS #1852: SkystarsRx Supreme Diversity RX won't join home network and stays as AP (3.0)

Pairan, 2022-09-20:
## Current Behavior
After flashing the Skystars RC Supreme diversity rx with the settings to join my wifi it boots up normal and works fine (Link to TX). But it wont enter the home network. 

Instead it starts up the AP mode and becomes visible in the `expresslrs` wifi ... 

## Steps to Reproduce
Simply flash the home network settings to the receiver and after that boot it without TX connection where it should enter the entered wifi.

## Your Environment

- TX hardware: BetaFPV Micro TX (unrelevant for this I guess)
- RX hardware: Skystars RX
- ExpressLRS version (TX & RX MUST MATCH): 3.0

Perhaps anybody could confirm this?

shafr, 2022-09-20:
Maybe you can flash firmware with Debug options on ? 
With that manual option checked and filled options from 
https://github.com/ExpressLRS/ExpressLRS/blob/master/src/user_defines.txt
it'll give more info to troubleshoot.

Pairan, 2022-09-21:
I will try it tonight and report back

Pairan, 2022-09-26:
didn't forget about it! just didn't had the chance so far! I will report back in this evening

shafr, 2022-09-26:
With mine ELRS Jumper NANO adapter I decided to order another one instead of trying to fix it. 
I have no idea what's going on there, I'll leave it for the better times, maybe it would be accidentally fixed with new release...

adamhun94, 2022-09-29:
@Pairan It is also happened to me with a new JHEMCU RX module. What I did was in AP mode on the web interface I selected the SSID from the dropdown list and entered the correct password. This way it is working fine! It looks like flashing an FW doesn't update the credentials in the EEPROM, because I tried flashing with a wrong password and still working fine.

pkendall64, 2022-10-05:
If you update the SSID/password in the UI then it overrides the value that is supplied at flash time, even if you flash a new firmware after updating in the UI. If you wish to go back to the flashed version of the SSID/password then on the options tab in the UI there is a link at the bottom "Reset all runtime options to defaults." that will clear all the custom options that have been entered in the UI.

adamhun94, 2022-10-05:
Thanks for the information! I guess this behavior is fine because you don't really change your SSID.

But what is the situation with brand new modules like mine? If I'm not mistaken, for me it looked like the initial flash actually did save the SSID, but not the password. On the UI it showed the SSID, but couldn't connect until I manually set the password in the browser. 
Did not try the "Reset all runtime options to defaults." though.

Pairan, 2022-10-05:
this pretty much describes my situation too ...

closed 2022-10-25 by pkendall64