ExpressLRS/ExpressLRS #1853: Frsky R9 Lite gets bricked when flashing from bootloader (3.0.0)

whinds, 2022-09-21:
Flashing device failed. 

## Steps to Reproduce

Use taranis x-lite and latest edgeTX version. The module had already been successfully flashed to ELRS 3.0.0 rc2 via the ST-Link.
Downloaded the new 3.0.0 firmware onto the SD card in the radio and then attempted to flash external module.

The progress bar during the "writing" stage will go to variable lengths. Sometimes the flashing attempt faisl immediately, sometimes it goes for a few seconds and then fails halfway through.

## Possible Solution (Not obligatory)

Might be some bootloader issues. There was something with v.0.5.3 working better than v.0.5.4 for the receivers. 

## Details

Frsky R9M Lite module does not flash via the bootloader and is bricked after attempting. Must be reflashed via bootlader to fix,

Sekilsgs2, 2022-09-26:
same here - new bootloader working very bad - 99% have error while flashing TX

mambovipi, 2022-11-07:
same here with otx 2.3.15. and x-lite r9 lite module. Not able to flash anymore :(
[image]

AlessandroAU, 2022-12-29:
Bootloader hasn't been changed in over a year. Xlite has always been iffy with flashing and no one ever found a root cause. Could also be something in recent OpenTX/EdgeTX changes. Are you able to try with another radio?

whinds, 2023-01-02:
Only tried on both an X-Lite original and X-lite Pro, so could be unique to them radios alone. My inkling is it is memory related because the progress bar can fail at different times. These are all the solutions I have tried, sometimes one works, and then doesn't. It is a very bizarre issue.
 1. Reflash the r9m_elrs_bl_.frk bootloader, then flash ELRS
 2. Reflash with repartitioner.bin but renamed to repartitioner.frk (this takes a few minutes to flash but has worked for me)
 3. Simply power cycle, turn radio on and off
 4. Plug in USB cable for more stable power supply
 5. Use a faster SD card in the radio
 6. Fully charge the radio batteries
 7. Try different versions of ELRS, then re-flash to latest
 8. Both OpenTX and EdgeTX exhibit this flashing problem, neither seems to work better than the other. However, in EdgeTX, changing the baudrate to a faster 946k instead of 400k was able to work sometimes. But then dont forget to change back to 400k or the LUA script wont work.

pkendall64, 2023-01-02:
The repartitioner is only for ESP based devices! It will not even run on an STM device.

whinds, 2023-01-02:
**Herme's voice** "That just raises further questions!"

It definitely finished with a "success" message. Then I was able to reflash with ELRS firmware successfully.

AlessandroAU, 2023-01-02:
The bootloader doesn't care what you upload it to, it'll happily take the file and push it into program space even if that doesn't make sense. Hence also why is doesn't make sense that ELRS 2.0 ELRS 3.0 could effect the bootloader.

If you've flashing .frk you're not even using our bootloader, you're invoking Frsky's stock bootloader.
the .elrs files are used with our bootloader. You only ever need to flash an .frk once, to install the elrs bootloader.

whinds, 2023-01-06:
Thanks for the explanation!
Since this seems to be isolated to X-lites and not other radios, could it be a wonky chip oscillator or some piece of hardware?

closed 2023-07-21 by github-actions[bot]