ExpressLRS/ExpressLRS #1854: Jumper AION 2.4TX nano stuck at bootup after downgrading from 3.0.0 to lower versions

matthewoots, 2022-09-22:
After flashing my board to `3.0.0`, and trying to downgrade to any version before that forces the module to be stuck in a startup loop. Would like to have some help to debug this.
Other issue on top of this is at `runtime` tab after connecting to the ELRS AP on `3.0.0`, there binding phrase is empty, and any `save and reboot` does not produce any change.
[image]

pkendall64, 2022-09-22:
The binding phrase in the 3.0 web UI is empty because ELRS does not store the phrase. It uses the phrase to generate the UID which is stored.
As you type the binding phrase you should see the UID changing underneath. And when you save and reboot and come back to the UI, the UID should be the same as when clicked "save and reboot". It should also have some text above the field saying the the UID was "Flashed" rather than "Not set".

As for downgrading, sometimes you need to fully erase the flash, and that is best done using esptool, which you can find on the web.

pkendall64, 2022-09-22:
Also, these questions are best asked on discord in the #help-and-support channel. The friendly folk there will help a lot quicker than using github issues.

matthewoots, 2022-09-27:
Thanks @pkendall64 the fix for anyone who will encounter this issue
```bash
git clone git@github.com:espressif/esptool.git --recurse-submodules
cd esptool
python3 esptool.py -p /dev/ttyUSB0 erase_flash
# Then use the configurator to flash again with UART, works like a charm
```

closed 2022-09-27 by matthewoots

kamarul-shah, 2024-07-21:
Hi Matthew..i have same problem with my module..can u help me 🥲