ExpressLRS/ExpressLRS #1855: DIY 2400 DUPLETX no AUTO_WIFI_ON_INTERVAL Functionality w/ESP32-WROOM-32

2xWhiskey, 2022-09-23:
## Current Behavior

     Using ExpressLRS Configurator v1.5.0, flashing DIY 2400 DUPLETX onto an ESP32-WROOM-32 & SX1280 E28 hardware combo does not support the AUTO_WIFI_ON_INTERVAL where flashing DIY 2400 TX ESP32 SX1280 E28 does.

## Steps to Reproduce

1.  Build module using a ESP32-WROOM-32 and a E28-2G4M27S SX1280

2.  Flash either ExpressLRS 2.5.1 or 3.0

3.  Power on module and allow to sit for set AUTO_WIFI_ON_INTERVAL time.

4.  Device does not auto create it's own wifi network as expected.

## Possible Solution (Not obligatory)

     Perhaps DIY DUPLETX does not support this hardware.  If so initiate documentation for DUPLETX to inform the layman of compatible/supported hardware.  DUPLETX documentation appears to be unpublished.

## Details

     I've also been unsuccessful at getting this hardware/firmware combination to successfully communicate with my Taranis X-Lite Pro via UART0 on the internal RF module connection, further suggesting this combination's incompatibility.  To verify the module's integrity, I successfully tested it by connecting it via the external module bay and using the DIY 2400 TX ESP32 SX1280 E28 firmware.

     I'm trying to replace my Taranis X-Lite Pro's internal RF module and retain the external module bay's functionality for other modules.  The radio is hard configured to disable the S.PORT line when the internal module is powered on, preventing use of the DIY 2400 TX ESP32 SX1280 E28 firmware.

     Don't think so, no clue what that is.

     Help me flick one more proverbial booger at FrSky for wantonly abandoning protocols and hardware by garnering more attention of people using ELRS on their hardware.  And the ergonomics of the X-Lite are just amazing for smaller handed thumbers such as my self.

## Your Environment

    All at my desk over a period of a few days troubleshooting, researching and pestering smarter people on Discord.

- TX hardware:  DIY 2400 TX ESP32 SX1280 E28, using a custom PCB to connect to the handset's internal RF bay connector
- RX hardware:  N/A, haven't gotten there yet
- Handset model:  FrSky Taranis X-Lite Pro
- OpenTX version (including nightly number):  EdgeTX 2.7.1
- ExpressLRS version (TX & RX MUST MATCH):  3.0.0 & 2.5.1
- Packet Rate:  115k & 400k
- Telemetry Ratio: unable
- user_defines:  AUTO_WIFI_ON_INTERVAL

iottrends, 2022-09-23:
have you looked into the pin mappings of your custom pcb.
since E28SX1280 is working the pin mapping. DIY_2400_TX_ESP32_E28.h .
However the pin mappings for **dupletx** is different:
#define GPIO_PIN_RCSIGNAL_RX    13
#define GPIO_PIN_RCSIGNAL_TX    13
#define GPIO_PIN_LED_WS2812     15
#define GPIO_PIN_FAN_EN         17

Compare above pins with the DIY_2400_TX_ESP32_E28.h
My guess is on flashing Dupletx your code is exiting before entering into bind WIFI, probably a pin and RF init issue.

2xWhiskey, 2022-09-23:
Thank you for your advice.  My code experience is next to nil so it will take me a while to research and decipher it for my understanding.

2xWhiskey, 2022-09-23:
Where can I find the aforementioned .h files?

pkendall64, 2022-09-23:
If you flash 3.0 using vscode/platformio and choose the "Unified_EXP32_2400_TX_via_UART" targets to flash, it will ask after building which device it is that you configuring, just press enter.
It will boot straight into wifi mode and when you go to the page you can configure the hardware layout.
If you look in the repository at the file in the directory [src/hardware/TX/](https://github.com/ExpressLRS/ExpressLRS/blob/master/src/hardware/TX/) you will see the files that are used when choosing a target for defining the layout.
Once you have your device configured, you can download the json file from the device by going to url /hardware.json

pkendall64, 2022-09-23:
I will move this to a discussion as it's not really an issue as such.