ExpressLRS/ExpressLRS #1862: Can't flash TX module, No USB connection, Wifi-Not enough space

stevempotter, 2022-09-29:
I am just getting started with Happymodel ES24TX Pro Micro I got today. It says it came with ELRS 2.0, so I am trying to flash 3.0 into it.
The Wifi flashing process got all the way to the end and then said "Not enough space". 
I read in another issue here with that error, that the advice was to flash with USB. But I cannot seem to get a connection.
On two different PCs with the CP210x VCP installed, no tones on inserting USB plug, no evidence of it in Device Manager. USB flash with configurator fails due to connection timeout.
On a Mac, I got this message at the end of the Build & Flash via USB plugged directly into the module (with dip switches 5 & 6 ON):

```
Serial port /dev/tty.SerialPort
Connecting...
Failed to get PID of a device on /dev/tty.SerialPort, using standard reset sequence.
....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
========================= [FAILED] Took 644.99 seconds =========================

Environment                                    Status    Duration
---------------------------------------------  --------  ------------
HappyModel_ES24TX_Pro_Series_2400_TX_via_UART  FAILED    00:10:44.985
==================== 1 failed, 0 succeeded in 00:10:44.985 ====================

```

The ES24TX is getting power (LED throbbing colors). Pins look perfect and device was reinserted a number of times in case it was a connection problem. My RadioMaster TX16S with latest EdgeTX can't seem to run any version (1, 2 or 3) of the ELRS Lua script. They all get stuck at the Connecting stage. I tried all the baud rates to no avail.
Is my module just broken? Or am I missing something?

Thanks for any advice.

pkendall64, 2022-09-29:
This is a question for the #help-and-support channel on discord rather than a github issue.
Could you please close this issue and ask there.

stevempotter, 2022-09-30:
OK, I have moved it. Thanks!

closed 2022-09-30 by stevempotter