ExpressLRS/ExpressLRS #1864: Script Syntax Error - elrsV3 Lua:489

skunkwurk, 2022-10-01:
This issue us easy to replicate.   When I try to enable WiFi using V3lua I get the syntax error in the image.  If I tried to run wifi using v2lau, it works fine.  Take a look at the pic, thanks in advance for the support. 

Setup:
- Radio - TX16S MK2
- Radio firmware: 2.8.0
- ES24TX Pro Micro ExpressLRS TX on ver 3.0.0 SM2G4 8aa1b0
- I downloaded the firmware and the lua using configurator v1.5
 
[image]

skunkwurk, 2022-10-01:
Has anyone had a chance to look at this?  I can provide a copy of the script or other details.

pkendall64, 2022-10-01:
You would be better to ask on discord in the #help-and-support channel as this is not a bug with the Lua script, as many users have this on their radios and have no problem. I'd suggest going to the web-ui on the TX module and clearing all settings and trying again.

skunkwurk, 2022-10-01:
Hi, thanks for the suggestion.  I'll try to ask in the support channel.  I have cleared the TX. It works fine on V2, it crashed when using V3 lua only.  It's def behaving like a bug.

skunkwurk, 2022-10-01:
Is this still the link?  https://discord.gg/expresslrs 

I keep getting invalid.

pkendall64, 2022-10-01:
Try this https://discord.gg/dS6ReFY seems discord is having issues with our custom link!

skunkwurk, 2022-10-01:
Thanks.  I was able to find the support group and we resolved the issue.  I had to
 bump up the baud rate.

closed 2022-10-01 by skunkwurk