ExpressLRS/ExpressLRS #1869: Can not build for Happymodel TX

cdjost, 2022-10-08:
When compiling with ELRS Configuration 1.5 I get the following error:

`
Building in release mode
PLATFORM : 'espressif32'
BUILD ENV: 'HAPPYMODEL_ES24TX_2400_TX_VIA_WIFI'
build version: 3.0.0 (8aa1b0) None

Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\common.cpp.o
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\dynpower.cpp.o
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\hal\hal_stm32.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\rxtx_common.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\tx_main.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
Generating partitions .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\partitions.bin
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib418\SPI\SPI.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
<command-line>:0:26: warning: missing terminating ' character
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\Esp32_i2s.c.o
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\HsbColor.cpp.o
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\HslColor.cpp.o
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\HtmlColor.cpp.o
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\HtmlColorNameStrings.cpp.o
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\HtmlColorNames.cpp.o
<command-line>:0:26: warning: missing terminating ' character
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\HtmlColorShortNames.cpp.o
<command-line>:0:26: warning: missing terminating ' character
<command-line>:0:26: warning: missing terminating ' character
<command-line>:0:26: warning: missing terminating ' character
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\NeoEsp32RmtMethod.cpp.o
<command-line>:0:26: warning: missing terminating ' character
<command-line>:0:26: warning: missing terminating ' character
*** [.pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\common.cpp.o] Error 1
Compiling .pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\lib056\NeoPixelBus\internal\NeoEsp8266DmaMethod.cpp.o
*** [.pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\dynpower.cpp.o] Error 1
*** [.pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\hal\hal_stm32.cpp.o] Error 1
*** [.pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\rxtx_common.cpp.o] Error 1
*** [.pio\build\HappyModel_ES24TX_2400_TX_via_WIFI\src\tx_main.cpp.o] Error 1
<command-line>:0:26: warning: missing terminating ' character
<command-line>:0:26: warning: missing terminating ' character
<command-line>:0:26: warning: missing terminating ' character
========================= [FAILED] Took 27.52 seconds =========================
`

pkendall64, 2022-10-24:
Without a full build log theres not much we can do here.

pkendall64, 2022-12-27:
From the error it looks like it's a problem with the platformio installation on your system.

closed 2022-12-27 by pkendall64

cdjost, 2022-12-30:
I tried it on a different machine with PopOs instead of Windows. Same Error. Also I cleared the platformio deps on both systems after the build and tried it again, no success.