ExpressLRS/ExpressLRS #1873: ArduinoCore/STM32 HardwareSerial is NOT ISR SAFE (R9MM lock-up)

alteman, 2022-10-10:
## Current Behavior

I'm experiencing receiver hangs on R9mm both with R9M and HappyModel TX modules and on multiple receivers.
Symptoms: 
Hang is pretty random, usually soon after powering up the receiver (about 5 min).
Both LEDs stay ON when it hangs.
FC OSD shows RXLOSS
TX telemetry shows no telemetry
TX power cycle doesn't clear this.
RX power cycle clears this.

I think I've narrowed it down to calling non-ISR safe HardwareSerial.write from ISR level.

ISR stack:
<Timer ISR>
hwTimer::callback
hwTimer::callbackTock
HWtimerCallbackTock 
crsfRCFrameAvailable
CRSF::sendRCFrameToFC
this->_dev->write
**HardwareSerial::write**

HardwareSerial::write is also called from 
loop
devicesUpdate
CRSF_device.timeout
CRSF::RXhandleUARTout

While getting the buffer is surrounded with noInterrupts/interrupts block, actual write isn't.

## Steps to Reproduce

Flash R9mm and any 900MHz TX to 3.0.0.
Leave the radio on.
Wait a while for rxloss/telemetry loss on the radio.
Optional: connect an FTDI adapter to collect some logs..

## Possible Solution (Not obligatory)

HardwareSerial is not ISR safe! 
Ideally all serial ops should be outside of ISR handlers.
See e.g. https://github.com/stm32duino/Arduino_Core_STM32/pull/1826
Also look at this nice line in HardwareSerial::write: 
```
  // If the output buffer is full, there's nothing for it other than to
  // wait for the interrupt handler to free space
  while (!availableForWrite()) {
    // nop, the interrupt handler will free up space for us
  }
```
[HardwareSerial.cpp#L510](https://github.com/stm32duino/Arduino_Core_STM32/blob/497ea5af8da63449365585dbcd05a09afd4ea00a/cores/arduino/HardwareSerial.cpp#L510)

## Details

Related issue: #1686 

Logs: 
[r9mm.log](https://github.com/ExpressLRS/ExpressLRS/files/9749498/r9mm.log)
[r9mm2.log](https://github.com/ExpressLRS/ExpressLRS/files/9749497/r9mm2.log)

## Your Environment

- TX hardware: HappyModel ES900TX
- RX hardware: R9mm
- Handset model: TX16S
- OpenTX version: EdgeTX 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.0.0 
- Packet Rate: 200Hz
- Telemetry Ratio: 1:64
- user_defines: Regulatory_Domain_EU_868 LOCK_ON_FIRST_CONNECTION

CapnBry, 2022-10-12:
Of course it is not safe to write to serial in an ISR, but DEBUG_LOG is only for debugging so just don't turn on all the debug log to fly?

alteman, 2022-10-20:
Fixed in 3.0.1

closed 2022-10-20 by alteman

MartinHugh, 2022-10-22:
> ## Your Environment
> * TX hardware: HappyModel ES900TX
> * RX hardware: R9mm
> * Handset model: TX16S
> * OpenTX version: EdgeTX 2.7.1
> * ExpressLRS version (TX & RX MUST MATCH): 3.0.0
> * Packet Rate: 200Hz
> * Telemetry Ratio: 1:64
> * user_defines: Regulatory_Domain_EU_868 LOCK_ON_FIRST_CONNECTION

Is this issue limited to 900/868Mhz or does it affect 2.4Ghz as well ?
Is 3.0.0 safe to fly with ?