ExpressLRS/ExpressLRS #1877: Wrong TX power is set on boot

SunjunKim, 2022-10-12:
## Current Behavior
When Max power is set lower than the `power_default` defined in the hardware descriptor, it's set to the default power on boot instead of the set Max power.

## Steps to Reproduce

1. Set the minimum power in Lua script
2. Power cycle
3. Observe the output power (using RF meter), which is set to `power_default` instead of the set power in the Lua.

## Your Environment
- TX hardware: ESP32 2.4g 1W module (min power 25 mW)
- RX hardware: Not relevant 
- Handset model: TX16S, TX12
- OpenTX version (including nightly number): >ETX 2.7
- ExpressLRS version (TX & RX MUST MATCH): master (bad7d1)
- Packet Rate: Not relevant 
- Telemetry Ratio: Not relevant 
- user_defines: Not relevant

MartinHugh, 2022-10-19:
Assuming I understand the scenario you are describing : 
I think this is expected behaviour when you turn the TX on *after* the RX.

The rationale, I imagine, is that if you have failsafed and then reestablishe connection, you want the power to be at the maximum permitted.  This is in contrast to when you turn the TX on *before* the RX when you will have minimum power.

SunjunKim, 2022-10-19:
> Assuming I understand the scenario you are describing : I think this is expected behaviour when you turn the TX on _after_ the RX.
> 
> The rationale, I imagine, is that if you have failsafed and then reestablish connection, you want the power to be at the maximum permitted. This is in contrast to when you turn the TX on _before_ the RX when you will have minimum power.

I appreciate your effort, but your comment is not relevant to this issue.

PR #1901 (2022-10-24): Fix power on boot being default power
Fixes #1877 

# Problem
If configured power is set to MIN_POWER, when the TX module is started it would select the default power level.

The problem was that the drivers would only check `pwrCurrent != pwrNew`, and set `pwrPending`.

1. On boot `pwrCurrent` is `SX12xx_MIN_POWER` set in the constructor and committed.
2. `POWERMGNT::init` sets power to default (not committed yet).
3. `ChangeRadioParams` calls `SetOutputPower` (with configured MIN_POWER) and the value is ignored because `pwrCurrent` == `pwrNew`!

Later the commit happens and sets the power to [... cut]
closed 2022-10-24 by JyeSmith