ExpressLRS/ExpressLRS #1905: elrs 3.0.0 for betafpv pwm reciver ch1 and ch2 is not working

lean0925, 2022-10-23:
## Current Behavior

I just update all tx and rx to elrs v3.0.0,when i pulg a servo to test it,ch1 and ch2 pwm output is not working,only ch3/ch4/ch5 is working.

## Steps to Reproduce

update tx and rx to elrs 3.0.0

## Details

I already try to flash another new pwm reciver still not working,and i also try to use anoter pwm target to flash it(DIY 2400 RX PWMP/Matek rx pwm),still only ch3/ch4/ch5 is working.
because tx v3.0.0 can't bind with rx v2.5.0,so this solution is not work.
fw 2.5.0 is working for all channel.
I tested my tx for another crsf based rx(happymodel EP1 rx ,betafpv lite 2400 rx),all channel worked.
because for my fix-wing plane need 4 ch for control , so this reciver is not working for me , and i get 3 of them.

## Your Environment

- TX hardware:betafpv 2400 Tx micro 1000mw,Jumper AION nano 2400 Tx
- RX hardware:betafpv pwm rx
- Handset model:Radio master tx16s,Jumper T-pro
- EdgeTX version :2.7.1
- ExpressLRS version (TX & RX MUST MATCH):3.0.0
- Packet Rate:500Hz
- Telemetry Ratio:1:128
- user_defines:
- DRegulatory_Domain_ISM_2400
-DHYBRID_SWITCHES_8
-DLOCK_ON_FIRST_CONNECTION
-DAUTO_WIFI_ON_INTERVAL="60"
-DFAST_SYNC

pkendall64, 2022-10-23:
Are you sure you don't also have DEBUG_LOG enabled? If you do then it uses PWM1 & 2 outputs for the debug serial port.

neko0925, 2022-10-25:
@pkendall64 
But my device option use standard mode and it don't have that DEBUG_LOG option,so i think i didn't enable that.
If i want to check the debug output,how can i do it? use FTDI usb-ttl connect pwm1 & pwm 2 and use serial monitor to ckeck output?

pkendall64, 2022-10-25:
> @pkendall64 But my device option use standard mode and it don't have that DEBUG_LOG option,so i think i didn't enable that. If i want to check the debug output,how can i do it? use FTDI usb-ttl connect pwm1 & pwm 2 and use serial monitor to check output?

Yes, exactly.

neko0925, 2022-10-26:
[image]
[image]
[image]

Is DEBUG_LOG default enable?because I try the config setting in screenshot ,but still only ch3 ch4 ch5 is working.

pkendall64, 2022-10-27:
Flashing using configurator will have the DEBUG_LOG off. Getting odd characters like that would suggest that something is coming out the pin, perhaps the PWM signal and the FTDI is trying to detect that as a serial stream.
Could you let the RX go into wifi mode and check the wifi page on the model tab and check the PWM pin mappings.
Also, it would probably be better to ask about this on discord in the #help-and-support channel as you get a much faster feedback and resolution to the problem.

neko0925, 2022-10-28:
Ok,thanks,I will close this issue and ask this problem in discord.