ExpressLRS/ExpressLRS #1912: Can't select 10mw or 25mW on BetaFPV 868MHz TX module

mha1, 2022-10-25:
## Current Behavior

There seems no option to select 10mw or 25mW on 868Mhz. Options only range from 100mW to 500mW. 10mW and 25mW are required for legal use of the in the EU.

Please allow for setting 10mW and 25mW at least for the 868MHz EU version.

## Steps to Reproduce

Try to select 10mW or 25mW

## Your Environment

- TX hardware: BetaFPV Micro Module 868MHz
- RX hardware: BetaFPV Nano 868MHz receiver
- Handset model: T16
- OpenTX version (including nightly number): Edge 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.0.1
- Packet Rate: tested with 50Hz and 200Hz, but seem to make no difference
- Telemetry Ratio: 1:2
- user_defines: 
Regulatory_Domain_EU_868MHz
Binding_Phrase
UART_Inverted
Unlock_higher_power
Arm_channel AUX1
Feature_opentx_sync
Hybrid_switches_8
Enable_Telemetry
tlm_report_interval_ms 100LU
Auto_Wifi_on_interval 60
BLE_HID_JOYSTICK
USE_DYNAMIC_POWER

pkendall64, 2022-10-25:
It's a hardware limitation. The amplifier it uses only goes down to 100mW.

closed 2022-10-25 by pkendall64

mha1, 2022-10-25:
Thanks for your reply. Do all 868 modules have this limitation?

pkendall64, 2022-10-25:
> Thanks for your reply. Do all 868 modules have this limitation?

No, it depends on the amplifier circuit the module uses.The Namimno Voyager goes down to 25mW

mha1, 2022-10-26:
> 

Sorry to bother you again. The Namimno Voyager seems to be no longer available. Do you know of any other 10mW/25mW module? Do you know if the Happmodel 868 is one of the capable ones?

pkendall64, 2022-10-26:
Yes, the HappyModel ES900 does go down to 10mW

mha1, 2022-10-26:
Great. Please allow me a real stupid question. Would it be possible to turn off the PA and just work with what the transceiver delivers?

pkendall64, 2022-10-26:
No, that's not possible as the PA is directly in the path of the Semtech chip and antenna port. Also, when you disable the PA via the control pins in shuts down completely. And having it at the minimum amplification level is what we use as the minimum anyway.