ExpressLRS/ExpressLRS #1927: Constant Failsafes in Ardupilot with V3.0.1!

AaryanL, 2022-11-02:
## Current Behaviour

After 6-7 min from start Ardupilot reports repeating short FS every couple of minutes!
[image]

## Steps to Reproduce

1. Just switch all on and wait )

## Details

1. Why is it the problem with ELRS - because if I use this setup and change only one thing - ELRS FW version to 2.5.1 - NO ANY FS any more, with exactly the same settings.
2. Problem disappears if Telemetry ratio is higher than 1:4 ( 1:2 and 1:4 - problem exists with Packet rate = 500 ).
3. Problem disappears if Packet rate <= 150 ( and Telemetry ratio = 1:2 ).
4. Absolutely ALL HW were replaced with the similar one and the problem reproduced.

## Your Environment

- Omnibus F4V3 + GPS at SERIAL3
- Arducopter 4.3.0 all default settings except of SERIAL1 SERIAL_PROTOCOL = 23, RC_OPTIONS = 800

- Taranis X9 lite
- OpenTX version 2.3.15
- ExpressLRS version 3.0.1 
- Matek RX with removed caps, link freq test passed

[image]
[image]
[image]

pkendall64, 2022-11-02:
Why have you raised another ticket? Please add this information to the other ticket and well close this one.

closed 2022-11-02 by pkendall64

AaryanL, 2022-11-02:
Because I've closed the old one. It has a lot of useless info concerning the issue. 

> Why have you raised another ticket? Please add this information to the other ticket and we'll close this one.

pkendall64, 2022-11-02:
It is better to leave the original ticket open and add this information to it as it shows the history of information gathering and things that have already been tried.

AaryanL, 2022-11-02:
Done.