ExpressLRS/ExpressLRS #1931: Expresslrs firmware build

AJAY-GANI, 2022-11-03:
Issue is i cant build 3.0.1 firmware for my quadkopters nano 2400rx please resolve the bug

pkendall64, 2022-11-03:
We have this fixed in the master branch.

AJAY-GANI, 2022-11-03:
Processing QuadKopters_NANO_RX_via_BetaflightPassthrough (platform: espressif8266@3.2.0; board: esp8285-8285; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
UID bytes: 88,19,143,157,143,139

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
!!!             ExpressLRS Warning Below             !!!
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
  Regulatory_Domain 2400 not compatible with RADIO_SX127X 
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

Exception: !!! Regulatory_Domain 2400 not compatible with RADIO_SX127X !!!:
  File "C:\Users\ganes\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 186:
    env.SConscript(item, exports="env")
  File "C:\Users\ganes\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\ganes\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\ganes\AppData\Roaming\ExpressLRS Configurator\firmwares\github\ExpressLRS\src\python\build_flags.py", line 147:
    print_error('Regulatory_Domain 2400 not compatible with RADIO_SX127X')
  File "C:\Users\ganes\AppData\Roaming\ExpressLRS Configurator\firmwares\github\ExpressLRS\src\python\build_flags.py", line 29:
    raise Exception('!!! %s !!!' % error)
========================== [FAILED] Took 5.54 seconds ==========================

Environment                                    Status    Duration
---------------------------------------------  --------  ------------
QuadKopters_NANO_RX_via_BetaflightPassthrough  FAILED    00:00:05.543
==================== 1 failed, 0 succeeded in 00:00:05.543 ====================
Result
Build error
An error has occured, see the above log for the exact error message. If you have not already done so, visit [Expresslrs.org](https://www.expresslrs.org/3.0/) and read the [Flashing Guide](https://www.expresslrs.org/3.0/quick-start/getting-started/) for your particular device as well as the [Troubleshooting Guide](https://www.expresslrs.org/3.0/quick-start/troubleshooting/#flashingupdating). If you are still having issues after reviewing the documentation, please copy the build logs above to an online paste site and post in the #help-and-support channel on the [ExpressLRS Discord](https://discord.gg/dS6ReFY) with a link to the logs and other relevant information like your device, which flashing method you were using, and what steps you have already taken to resolve the issue.

closed 2022-11-03 by pkendall64