ExpressLRS/ExpressLRS #1932: Packet rate not updating in TBS Mambo

deeph46, 2022-11-03:
Hello,

Have issue with ELRS in TBS mambo with BetaFPV TX Nano module. The packet rate om top right side not changing when I change Packet Rate setting.

Module running 3.0.1 and Mambo running latest FW 1.34.

How to fix this?

AlessandroAU, 2022-11-04:
What packet rate are you setting and what is the reading?

deeph46, 2022-11-04:
Top right corner showing 0/250, I chnaged into 150Hz got the same reading (0/250).

wvarty, 2022-11-11:
250hz is the fixed mixer rate in old versions of OpenTX, off which FreedomTX is based. I assume you are likely running an old version of FreedomTX, and an update should fix.

deeph46, 2022-11-13:
I run latest Beta FreedomTX right now. Which side for the update fix, from FreedomTX or ELRS?

wvarty, 2022-11-13:
FreedomTX. Would not surprise me if TBS have broken it again. Woundn't be the first time.

deeph46, 2022-11-13:
I see, so for a while is it save to fly with missmatch packet rate?

AlessandroAU, 2022-11-22:
@deeph46 Currently investigating, version 1.30 appears to work on my Tango2.

deeph46, 2022-11-22:
> @deeph46 Currently investigating, version 1.30 appears to work on my Tango2.

Ooh, yes. 1.30 actually work, forgot to mention it before. I now in 1.30, but I have my quad setup for Tracer either. I will try back to ELRS later.

closed 2022-12-27 by pkendall64