ExpressLRS/ExpressLRS #1941: Happymodel EP1 Dual pulled into weird state when on UART1 of Mamba DJI F405 Mini Mk2

rabbitambulance-ra, 2022-11-11:
## Current Behavior

_Filing per Sunjun:_ [https://discord.com/channels/596350022191415318/950918956779208754/1040430473576448020](https://discord.com/channels/596350022191415318/950918956779208754/1040430473576448020)

Happymodel EP1 Dual does not communicate with the Mamba DJI F405 Mini Mk2 flight controller while wired to UART1 (and almost certainly other Mamba F405 FCs, and possibly other FCs with an F.Port pad) when powered from USB prior to vbatt

## Steps to Reproduce

1. Wire Happymodel EP1 Dual to RX1 / TX1 of a Mamba F405 FC
2. Power from USB port
3. TX will communicate with RX and be visible in the lua on the TX
4. BF will not see data from the receiver despite Serial RX being enabled on UART1, and Serial/CRSF selected as the mode and protocol. There will be no stick movements and no data will be received from the RX when rxpassthrough is enabled for UART1 in the CLI

## Possible Workaround

Plug vbatt in before USB. RX will boot correctly, and communicate with the flight controller without issues. With no configuration changes whatsoever, all channel inputs will be registered on the BF Receiver tab.

The most straightforward workaround is of course using a different UART, but that's problematic for GPS builds with these FCs due to UART2 being DJI plug, and UART3 having only TX3 broken out.

## Details
Please see attached FC diagram in case it is helpful:

[image]

## Your Environment

- TX hardware: N/A, since it is reproducible with all TXs I own, including DIY slimmer, Jumper AION TX, HM ES24TX Slim Pro, and Jumper T-Pro internal
- RX hardware: Happy Model EP1 Dual (five different EP1 Dual RXs tested, and all behave the same)
- Handset model: Jumper T-Pro, TBS Tango 2
- OpenTX version (including nightly number): Edge TX 2.8 nightly (RC2ish vintage)
- ExpressLRS version (TX & RX MUST MATCH): 3.0.1
- Packet Rate: all packet rates
- Telemetry Ratio: all telem ratios
- user_defines:
```
-DRegulatory_Domain_ISM_2400
-DMY_BINDING_PHRASE="RabbitAmbulance"
-DLOCK_ON_FIRST_CONNECTION
-DAUTO_WIFI_ON_INTERVAL="60"
-DFAST_SYNC
```

rabbitambulance-ra, 2022-11-11:
Closing this for now while I sanity check a few things coming out of the chat on discord

closed 2022-11-11 by rabbitambulance-ra

jawz101, 2022-11-15:
please keep this open.  I have a [Diatone Mamba F722 MK4 Mini](https://www.diatone.us/products/mb-mk4-f722-mini-fc) FC with 2 ExpressLRS receivers with this problem.  Flashed 3.0.1 on both receivers.  My Radiomaster TX16s MKII can communicate with both receivers, however neither can talk w/ either Betaflight or iNav on the flight controller.  The Happymodel is currently connected and I tried wiring it to UART1 (TX1/RX1) and  then UART5 (TX5/RX5) [per the instructions on the wiki](https://www.expresslrs.org/2.0/quick-start/receivers/hmep2400/) and I still get a solid light as if the receiver is still in bootloader mode (I guess.)  I'm afraid to keep soldering and unsoldering this or buying more receivers to find some receiver that is supposed to work when everything I see says it should.

https://cdn.shopifycdn.net/s/files/1/0027/2708/4144/files/MK4_F722_MINI_42688P_-01.jpg

[Happymodel 2.4g ExpressLRS ELRS EP1 RX](https://www.happymodel.cn/index.php/2021/04/10/happymodel-2-4g-expresslrs-elrs-nano-series-receiver-module-pp-rx-ep1-rx-ep2-rx/)

[image]

[0.4g Flywoo 2.4G ExpressLRS ELRS EL24P](https://flywoo.net/products/0-4g-flywoo-2-4g-expresslrs-elrs-el24e-el24p?variant=40438891708611)

[image]

rabbitambulance-ra, 2022-11-15:
Come to the Discord #help-and-support: [https://discord.gg/WtJqU39j](https://discord.gg/WtJqU39j), @jawz101. We'll get you sorted out. 
My issues is specific to the EP1 Dual, and is different from your issue.