ExpressLRS/ExpressLRS #1944: Joystick BLE stuck when enabled from lua script

kvark85, 2022-11-13:
## Current Behavior

After turning on the BLE joystick using the lua script, the program stucks (when the BLE Joystick is activated with the joystick on the ExpressLRS module, everything works fine).

## Steps to Reproduce

1. Go to TOOL -> ExpressLRS

2. Click on the **BLE Joystick**

3. The program will stuck
[image]

## Possible Solution (Not obligatory)

## Details

The same issues occur when I try to enable wifi from a lua script.
1. Go to TOOL -> ExpressLRS -> WiFi Connectivity
2. Click on the **Enable WiFi**

## Your Environment

- TX hardware: BETAFPV ExpressLRS 2.4 1000mW 
- RX hardware:
- Handset model: Radiomaster TX12
- OpenTX version (including nightly number): 2.3.15-otx (1004ef7c)
- ExpressLRS version (TX & RX MUST MATCH): 3.0.1
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2022-11-13:
If the [X] thing is changing then it's not stuck, it's working as expected.
If you press the RTN key it will exit BLE mode. If you press and hold the RTN key it will exit the LUA, leaving the BLE running.

If the [X] is frozen then we have an issue.

kvark85, 2022-11-13:
Sorry, it it was not obvious for me. Thanks, then it works as expected.

closed 2022-11-13 by pkendall64