ExpressLRS/ExpressLRS #1945: Update to 3.0.1 via BF Passthrough bricks EP1/2 RX

Xedos9er, 2022-11-13:
## Current Behavior
Flashing Happymodle EP1 or EP2 to 3.0.1 via Betaflight passthrough using ELRS configurator lead to bricked unit

## Steps to Reproduce

1. Choose 3.0.1 Update for Happymodel EP 2400 RX in Configurator

2. start flash

3. after flash done, power cycle

4. RX starts blicking slowly and not responding

## Possible Solution (Not obligatory)

Short Bootloader pins and reflash via configurator with 3.0.0 firmware.
Then use WIFI update to update on 3.0.1 firmware

## Details

3.0.0 seems to be fine via BF Passthrough, 3.0.1 is broken.
WIFI update works fine.
Other RX not affected so far (tested Matek and BETAFPV SuperD

## Your Environment

- TX hardware: ES24 PRo
- RX hardware: EP1
- Handset model: Radiomaster TX16s
- OpenTX version (including nightly number) EDGE TX 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.0.1
- Packet Rate: 250hz
- Telemetry Ratio: 1:16
- user_defines: none

evilC, 2022-12-01:
I am seeing EP2s get bricked when I try to flash using WiFi.
Am able to recover OK by BF passthrough / boot pad soldering, and then flash to 3.0.0
It happened to both EP2s that I tried.

My process was (Both old EP2s that shipped with ELRS 1.x)

1. Flash 2.5.1 (Because I can't flash to 3.x straight away, get the "Not enough space" error)
2. Flash 3.0.1 - Update seems to work OK, but when it reboots, the LEDs flash at a rate faster than the normal "Waiting for connection" rate but slower than the "In bind mode" rate

jabacz, 2022-12-01:
Workarround: 
Before flash
1) Disable BetaFlight telemetry -> reboot
2) Do NOT power on the transmitter. E.g. RX is without ELRS link to TX
Flash using passthrough

After flash
1) Enable back BF telemetry -> reboot
2) Turn on the TX

jabacz, 2022-12-01:
Workaround: 
Before flash
1) Disable BetaFlight telemetry -> reboot
2) Do NOT power on the transmitter. E.g. RX is without ELRS link to TX
Flash using passthrough
1) Enable back BF telemetry -> reboot
2) Turn on the TX

closed 2023-04-14 by github-actions[bot]