ExpressLRS/ExpressLRS #1946: ELRS LUA: Script Syntax Error

Killerkombo27, 2022-11-13:
## Current Behavior

Whenever I enable WiFi on my external module "Script syntax error" appears on my screen with one of these two messages underneath: 

?:0: attempt to index field'?'(a nil value)

TOOLS/elrsV3.lua:489 attempt to index field'?'(a nil value)

## Steps to Reproduce

1. Download Lua script over WiFi for HM ES24TX Pro Series 2400 TX under the device category Happymodel 2.4 GHz(use release 3.0.1)

2. Enable WiFi for the external module on the radio

3. Wait a few moments and the messages will appear on the radio screen

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: Happymodel ES24TX Pro 2.4GHz Micro ExpressLRS TX Module
- RX hardware: Happymodel ExpressLRS Nano 2.4GHz EP2 RX
- Handset model: TX16SE
- EdgeTX version (including nightly number): 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.0.1
- Packet Rate: 500Hz @ 100mW
- Telemetry Ratio: 1:128
- user_defines:

Darylc46, 2022-11-16:
Was there a fix for this?  Having the same problem!

Killerkombo27, 2022-11-16:
> Was there a fix for this?  Having the same problem!

Sorry should've wrote the solution but I found that if I upped the bitrate in the model selection this error didn't show up. You can find the bitrate in the place where you choose internal or external module and all that.

Darylc46, 2022-11-16:
no worries glad you sorted yours! 
sorry i have a qx7s and would it be under Buadrate? mine is currently at 115k

Killerkombo27, 2022-11-16:
> no worries glad you sorted yours! 
> 
> sorry i have a qx7s and would it be under Buadrate? mine is currently at 115k

Yes it should be under Buadrate and when I set mine to 921k it worked for me

Darylc46, 2022-11-16:
It worked! I no longer have the error, but the Wi-Fi seems to drop out and it won't let me update the module

Killerkombo27, 2022-11-16:
> It worked! I no longer have the error, but the Wi-Fi seems to drop out and it won't let me update the module 

All I can think of is try lowering the Baudrate above 115k and below 921k and see if that work. If that doesn't work then I would try redownloading the LUA script and rebuilding the firmware and set that up on the module then I would try updating it again at 921k and the other Buadrate above 115k

pkendall64, 2022-11-16:
The wifi not staying connected, depends on if you have configured home wifi credentials or not. 
For home wifi connection, the module needs to be fairly close to the router/access point.
If using the modules access point, then the computer/phone needs to be close to the module.

closed 2022-11-17 by Killerkombo27