ExpressLRS/ExpressLRS #1950: Not able to Flash QK NANO RX 2400 with ELRS V3

satyagupta, 2022-11-16:
Not able to flash QK Nano Receivers with target QK NANO RX with ELRS v3.0

Try to build using elrs configurator with Target QK NANO RX (QuadKopters > QK NANO Rx)

[image]

Works fine with V2.0 but not with V3.0

pkendall64, 2022-11-16:
We've fixed this and it will be released in 3.1, which should hopefully be very soon.

satyagupta, 2022-11-16:
Thanks @pkendall64 i hope it comes out sooner

Users are not able to flash thinking its a hardware issue and moving to different hardware :(

pkendall64, 2022-11-16:
> Thanks @pkendall64 i hope it comes out sooner
> 
> Users are not able to flash thinking its a hardware issue and moving to different hardware :(

We are talking about this week. We noticed this the other day and I wrote a new verifier that runs on the github build so this should not happen again.

satyagupta, 2022-11-16:
Excellent, you guys are best!

satyagupta, 2022-11-28:
Any update on this guys?!

deadbytefpv, 2022-11-28:
> Any update on this guys?!

https://github.com/ExpressLRS/ExpressLRS/pull/1922

I think this can be closed now.. :D

closed 2022-12-19 by JyeSmith