ExpressLRS/ExpressLRS #1952: LUA Script "Loading..." 2.5 minutes on radiomaster tx12 mk2 elrs v3.0.x

AutomationEngineer, 2022-11-17:
## Current Behavior

LUA Script "Loading..." 2.5 minutes and more on radiomaster tx12 mk2 elrs with internal module flashed with version 3.0.x thowever it starts instantly on this hw when flashed with version 2.5.1. Lua script is v3.

## Steps to Reproduce

1. Pick up radiomaster tx12 mk2 elrs

2. Flash internal tx module with version 3.0.0 or 3.0.1

3. Run ElrsV3.lua script

4. Wait couple minutes for it to start, looking on "Loading..." screen

## Details

I need version 3.0.x because my receiver supports only version 3.0.0 and higher (Matek 2.4GHz ELRS-R24-P6 PWM (DIY 2.4 GHz / DIY 2400 RX PWMP EX))

## Your Environment

- TX hardware: radiomaster tx12 mk2 elrs
- RX hardware: Matek 2.4GHz ELRS-R24-P6 PWM (DIY 2.4 GHz / DIY 2400 RX PWMP EX)
- Handset model: radiomaster tx12 mk2 elrs
- EdgeTX version 2.7.1 and 2.8.0 rc4
- ExpressLRS version 3.0.0 or 3.0.1
- Packet Rate: 500
- Telemetry Ratio: 1/4
- user_defines: don't know

pkendall64, 2022-11-17:
Please ask in the discord help-and-support channel, you'll get a much faster resolution there, as there are plenty of folk willing and able to help.

ELongRange, 2022-11-18:
Can confirm, this is a bug. TX12 MkII radio internal module does not like ELRS 3.0.0 or 3.0.1 firmware.
FW 2.5.1 runs just fine including wifi and BLE joystick.

Tested (again and again) with EdgeTX v2.7.0 v2.7.1 v2.8.0 RC1-RC4, using correct bootloader and ElrsV3.lua script.

The flashing of the FW itself works everytime at the same speed and without hiccups for all ELRS firmwares in ExpressLRS Configurator v1.5.2 using EdgeTxPassthrough.

With ELRS 3.0.0 and 3.0.1 firmware the LUA script loads extremely slow with baudrate set at 400k. Lower baudrates are even slower.
Higher baudrates than 400k makes loading faster and appear to complete loading the script, but still it's not possible to change any values. Changing a value with the scroll wheel works but after entering the value reverts to whatever it was at before changing. 

As it is the radio is unusable with newer receivers running only FW 3.0.0 and higher.

pkendall64, 2022-11-18:
I just tested on my TX12S MKII radio and it loads very quickly.
Using V3.0.1 firmware on EdgeTX 2.8.0-RC4, tested with 400k, 1.87M and 5.25M and all load very quickly.
The sample mode setting in where the baud rate is is set to "Normal".
@ELongRange I would suggest you ask on the discord server as well.

ELongRange, 2022-11-18:
Thanks for testing. I was hoping for a bug in the firmware but this is something else and a bit weird because the 2.5.1 FW is flawless.
Just tested again and same results, this is definitely not 'user error'. I will ask on the discord.

ELongRange, 2022-11-18:
**FIXED!!** Found the solution by searching the help-and-support section on the Discord server.
**In ExpressLRS Configurator go to the support section and click 'CLEAR PLATFORMIO DEPENDENCIES' and also 'CLEAR FIRMWARE FILES'**
After that reflash FW and all should be well, I'm somewhat ecstatic after a week of frustation..
After testing everything seems allright with the new FWs, including WIFI, BLE and changing settings.

thanks [pkendall64](https://github.com/pkendall64)

AutomationEngineer, 2022-11-19:
Other person on discord also don't have any issues with 3.0.x and tx 12 on any baudrate set. I also found that sometimes with 400k it loads faster, sometimes slower. On other rates it is allways slower. And on 5.25M it never boots. On 2.5.1 it allways boot fast on any baudrates inclluding 5.25. Could it be related to particular hw version don't like new fw? Mine is v2.9

ELongRange, 2022-11-19:
Hardware version for the elrs module is V2.9 
Hardware version for the radio is V2.3

[image]
[image]

Aaand the new v3.1 is even worse, not loading at all at any baudrate, 2.5.1 is fine as usual. Wasted half a day again. I'm seriously souring on the whole ELRS thing, the issue is not common but it is time somenone will aknowledge that there is a problem instead of passing it off as user error.
Radiomaster refuses to even answer my questions (guess I have to be a youtuber) and the Droneshop.nl 'support' advises me to check if the radio is set to CRSF... (really?)
It's just flaky and I would never trust this system on an expensive plane besides it being dangerous and irresponsible.
Back to the old and trustworthy I guess. Maybe in a few years...

ELongRange, 2022-11-19:
I'm not giving up, deleting the elrsV3.lua**c** from the radio after flashing and loading speed is normal again.
But still no luck changing any settings. See youtube video I made:

https://www.youtube.com/watch?v=dV2NY9VdIVQ

Interestingly changing the 'Switch Mode' gives an error 'not while connected' and the number 32. See video. This error behaviour is consistent over all firmwares from 3.0 and up:

[image]

ELongRange, 2022-11-19:
**_Anyone testing this_**:
It is not about loading speed it's about not being able to change values (see my video)
So don't just test the loading speed, try and change settings in the LUA script.

AutomationEngineer, 2022-11-19:
I have no issues with changing values, once script loaded, I can change values.
And I have found a commit which breaks loading for me: https://github.com/ExpressLRS/ExpressLRS/commit/2c2114a1deef16912eab828651f25f1eb31e9de5
Unfortunately it is so hudge to locate exact cause..

ELongRange, 2022-11-19:
Well that makes two of us. I've been at it for days, today all day on Discord (where the DEVs have been super helpful) multiple FWs failing and no solution in sight.
This junk is going back and I'm buying a TX16S

lesharodin, 2022-11-30:
Had similar issues but with external elrs on old tx12
https://github.com/ExpressLRS/ExpressLRS/issues/1848
There was suggestion by @pkendall64 to do full erase on flash, but I never find that option anywhere.
So still using 2.5.1

AutomationEngineer, 2022-11-30:
@lesharodin, thank you for your comment, it helped a lot!

With the mentioned [tool](https://pkendall64.github.io/elrs-web-flasher/), I was able to erase flash and burn a 3.0.1 furmware. Despite it didn't work directly after this, it worked after flashing again from ExpressLRS Configurator.

No delay now and any baudrate work!

Thank everybody for help!

ELongRange, 2022-11-30:
@AutomationEngineer, YEAH! it finally works. Exactly like you said. Didn't take the first time but a second flash in the configurator did it. Thank you for persevering and the solution.

ELongRange, 2022-11-30:
@lesharodin, try it, it really works for the new 3.0 and 3.0.1 FWs!

1. Make sure the STM32 VirtualComPort driver for Windows is installed. Get is from here:
https://quadmeup.com/stm32-virtualcomport-driver-for-windows/
2. Turn on TX12, connect to USB port on windows computer and on TX12 menu choose 'USB Serial (USB)'
3. Go to site https://pkendall64.github.io/elrs-web-flasher/
4. Flash firmware according to screen caps Capture1.PNG, Capture2.PNG enCapture3.PNG, mind the orange marks. make sure to have select erase before flashing.
5. Reflash new firmware in ExpressLRS Configurator like AutomationEngineer said. Done.

Many thanks to **PKendall** pointing me to his web flasher before but I couldn't get it to work then for some reason,this time it worked great and all is well.
[image]
[image]
[image]

lesharodin, 2022-12-07:
Devs, shouldn't we have erase flash option on desktop configurator?
Anyone who flash 3.x from 2.x via uart without repartioner would face this strange issues. I have around 10 people in our community affected by that bug.

closed 2022-12-09 by pkendall64

drainbeast215, 2023-01-03:
Did you ever get that fixed?? I'm having the same problem

drainbeast215, 2023-01-03:
Did you ever get that fixed?? I'm having the same problem now

lesharodin, 2023-01-03:
> Did you ever get that fixed?? I'm having the same problem

Read this thread, solution: https://github.com/ExpressLRS/ExpressLRS/issues/1952#issuecomment-1332843863

rdavyd, 2023-03-22:
@drainbeast215 Try checking port speed on RC radio control for CRSF. I had mine set to 115k instead of 400k and was experiencing the same issue.