ExpressLRS/ExpressLRS #1955: LUA Script error on PWM Receiver

gjhaynes, 2022-11-20:
## Current Behavior

Viewing the Rx parameters of a Matek R24-P6 on a TX16S using the ExpressLRS script results in a script error
## Steps to Reproduce

1. Flash the Rx with ELRS v3.1.0

2. Execute the ExpressLRS LUA script (v3) on the TX16S

3. Choose "Other Devices and select the ELRS+PWM 2400RX

4. At the point where Output Mapping should be displayed, a script error occurs: "bad argument \#.1 to 'char'(number expected, got nil)

## Possible Solution (Not obligatory)

## Details

The problem appears related to the new feature in ELRS 3.1.0 that displays channel mapping for PWM receivers

Unable to view Rx properties or edit channel mapping from the handset using the ELRS script

No changes to boilerplate

## Your Environment

- TX hardware: Happymodel ES24TX Pro 1000mW Micro ExpressLRS 2.4GHz ELRS TX
- RX hardware: Matek R24-P6
- Handset model: RadioMaster TX16S
- EdgeTX version (including nightly number): 2.8 RC4
- ExpressLRS version (TX & RX MUST MATCH): 3.1.0
- Packet Rate: 400K (error persists if rate is changed)
- Telemetry Ratio: 1:16 (error persists if ratio is changed)
- user_defines:

paulkidd68, 2022-11-20:
I'm Getting the same error on a Radiomaster TX16S with a BetaFPV PWM receiver too.

ajjjjjjjj, 2022-11-21:
It might be related - I found out that "Output mode" menu item is not sending last 2 bytes in packet so unit "\0" is not existent.

pkendall64, 2022-11-21:
Hi there, we have a PR in place if you'd like to test and provide feedback. #1962

jurgelenas, 2022-11-27:
@pkendall64 should we close this? It was fie din #1962

closed 2022-12-09 by pkendall64