ExpressLRS/ExpressLRS #1956: Update to 3.1.0 almost bricks Jumper T-Pro Internal module

XenithBoroda, 2022-11-20:
## Current Behavior

After update Jumper T-Pro Internal module to 3.1.0 LUA script stops working (tested on wi-fi update and edgetxpassthrugh)
After timeout internal module brings up wifi. Browser opens http://10.0.0.1 with hardware, not index.html
All fields in hardware are empty. Target configuration JSON file nor manual fill don't save.
Update over wifi not possible anymore

## Steps to Reproduce

Update module to version 3.1.0

1.

2.

3.

4.

## Possible Solution (Not obligatory)

Downgrade to 3.0.1 via edgetxpassthrough

## Details

## Your Environment

- TX hardware: Jumper T-Pro Internal ELRS
- Handset model: Jumper T-Pro
- OpenTX version (including nightly number): EdgeTX 2.8.0-rc4
- ExpressLRS version (TX & RX MUST MATCH): 3.1.0
- Packet Rate: 250
- Telemetry Ratio: std
- user_defines:

deadbytefpv, 2022-11-20:
From what version is it?
i.e. What version does the internal module have prior to flashing to 3.1.0?

bigbear2001, 2022-11-20:
I have the same
from 3.0.1 to 3.1.0

deadbytefpv, 2022-11-20:
Can confirm this on mine.
The Happymodel EP Dual, and possibly the BetaFPV SuperD is affected as well.. 

This is still under investigation by the Dev Team.

Would advice against flashing 3.1.0 to these devices for now until a hotfix is released.

XenithBoroda, 2022-11-20:
> From what version is it? i.e. What version does the internal module have prior to flashing to 3.1.0?

From 3.0.1

sjebd, 2022-11-20:
Same on my Jumper T Pro...

Tried to update from 3.0.1...

Soldering the wire to the right shoulderpad was needed..(as bootloader button).
Had then to "unbrick" via EdgeTX Passtrough and downgrade again...

mubispilvas, 2022-11-20:
The same with Jumper AION 2.4TX NANO, flashed from 3.0.1

artnascimento, 2022-11-22:
Please someone help me! Anyway I try to update the error, the wi-fi does not appear and neither via Edgetx Passtrough (using the bootloader button) It did not work. I don't know if I'm pushing at the wrong time. I've tried several times

JoeriCapens, 2022-11-22:
I had the same issue on my Jumper T-Lite V2. The #1961 PR merged into the 3.x.x branch fixed it (even when building on my Debian PC).

closed 2022-12-09 by pkendall64