ExpressLRS/ExpressLRS #1958: BETA FPV 2.4GHz Nano TX Update to 3.1.0 CE_LBT

stevewitwitty, 2022-11-20:
## Current Behavior

Tried to update the module to 3.1.0 CE-LBT but Express LRS Lua script stays on loading screen and does not bind to the reciever. Get telemetry lost telemetry recovered warnings every few seconds.
Both receiver and transmitter updated to 3.1.0.
Latest V3 version of Lua script installed on SD card.
Changed transmitter back to 3.0.1 and now its fine. Only has the problem when TX module is updated to 3.1.0

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: BetaFPV 2.4GHz Nano module
- RX hardware: BetaFPV 2400 Nano RX
- Handset model: FrSky X9 LITE
- OpenTX version (including nightly number) N/A
- Edge TX version: 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.1.0
- Packet Rate: 250Hz
- Telemetry Ratio: Std (1:64)
- user_defines:

stevewitwitty, 2022-11-20:
I have another receiver on another model the IFlight 2400 RX also updated to 3.1.0 and have the same problem when the TX is updated to 3.1.0. Must be a problem with the TX module. Reverting the TX to 3.0.1 fixes the issue.

jurgelenas, 2022-11-27:
@pkendall64 should we close this?

pkendall64, 2022-11-28:
> @pkendall64 should we close this?

I marked it as v3.1 so when release 3.1.1 I can close this and others

PSi86, 2022-12-09:
Not sure if it connected or not but after flashing 3.1.1 to the internal zorro elrs module there is the same behavior. Lua not loading. Switching from 3,x MBaud to 1,x brings back the lua.

stevewitwitty, 2022-12-09:
Sadly this issue still happens with latest 3.1.1 update. Issue is not sorted

closed 2022-12-09 by pkendall64