ExpressLRS/ExpressLRS #1959: Can not flash 3.0.0 or 3.0.1 on Matek 2400 RX R24D - Not enough space  error

BrunoB99, 2022-11-21:
## Current Behavior

[image]

## Steps to Reproduce
- Build firmware 3.0.0 or 3.0.1 selecting device "Matek 2400 RX R24D"
- Connect to Receiver's web page and force flash the firmware. I need to force the flash because the device name has changed from 2.x to 3.x (Note from Matek's site: for ELRS-R24-D, if update from 2.x to 3.x, Pls select target MATEK 2400 RX R24D and click on Force Flash.)
- Get an error "Not enough space" when flashing.

If I build 3.0.0 or 3.0.1 selecting device "Matek 2400 RX" that works fine and the file is the same size.
If I build 2.5.1 selecting "use diversity" that also works.

pkendall64, 2022-11-21:
It says in the release notes what to do in this situation! Please read them and use the repartitioner.

BrunoB99, 2022-11-21:
Hi Paul,

Thanks for the quick response. I did read the release notes but I did not think it applied for two reasons:

  1.  I am trying to flash a receiver and the release notes talk about TX module.
  2.  I was able to flash 3.0.0 and 3.0.1 if I build selecting the device "Matek 2400 RX", but I get the error when I build selecting the device "Matek 2400 RX R24D"

I did flash the file "repartitioner.bin" and now I think my receiver is blicked. The led is steady on and I can not coonnect to it.

Bruno

SunjunKim, 2022-11-21:
> It says in the release notes what to do in this situation! Please read them and use the repartitioner.

Isn't the repartitioner only for a TX?

deadbytefpv, 2022-11-21:
@BrunoB99 
Flash the receiver to 2.5.1 FIRST, then flash it to 3.x.. 
There's two other flashing methods as well: via Passthrough and via FTDI

pkendall64, 2022-11-21:
> Hi Paul, Thanks for the quick response. I did read the release notes but I did not think it applied for two reasons: 1. I am trying to flash a receiver and the release notes talk about TX module. 2. [...]

Yeah, sorry about that. I read and replied too quickly. There was so much going on yesterday tracking issues down.

BrunoB99, 2022-11-24:
I was able to build and flash from the configurator using the UART method. We csn close this issue now. Thanks for the help.

closed 2022-11-24 by BrunoB99

cbiphu4ok, 2023-01-06:
definitely don't use "repartitioner" for RX. I just tried it and now my transmitter is broken :(

pkendall64, 2023-01-07:
> definitely don't use "repartitioner" for RX. I just tried it and now my transmitter is broken :(

It won't be bricked, you will just need to flash via UART (FTDI or passthrough) but putting the RX into bootloader mode by holding the button or shorting the boot pads.

cbiphu4ok, 2023-01-07:
ELRS Nano Receiver 2.4G BetaFPV with a Button

======== PASSTHROUGH DONE ========
======== RESET TO BOOTLOADER ========
  * Using full duplex (CRSF)
Cannot detect RX target, blindly flashing!
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM3
Uploading .pio\build\BETAFPV_Nano_2400_RX_via_BetaflightPassthrough\firmware.bin
esptool.py v4.2.1
Serial port COM3
WARNING: Pre-connection option "no_reset" was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
Connecting......................................

A fatal error occurred: Failed to connect to ESP8266: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
========================= [FAILED] Took 27.13 seconds =========================

pkendall64, 2023-01-07:
The best place to ask for this sort of help is on the discord server in the #help-and-support channel.
But you must make sure the button is held down when you power the RX. So the LED should be on solidly.
If you need to plug in a LiPo to power the RX then, again, hold the button down while plugging in the LiPo. If you don't need a LiPo to power the RX then hold the button down when inserting the USB cable.

Also, since you are flashing via BetaflightPassthrough, I would assume that you have the RX working on this flight controller before and was getting betaflight telemetry back on the radio! If not then it won't work and you may need to disconnect it from the FC and use an FTDI adapter.