ExpressLRS/ExpressLRS #1960: ExpressLRS v3.0.1 sends different rc data without user command which is causing disarm and motor glitch momentarily

ckflight, 2022-11-21:
Today i have updated my HappyModel ES24TX, HappyModel EP1 2400 RX to 3.0.1 and Taranis QX7 to latest OpenTX version. 

Besides having "not enough space" error during wifi update (i have updated by using usb interface), receiver gets different data sometimes. I am not touching to Taranis QX7 and it is standing still but it receives different values. 

The parsed crsf data output can be seen from the first picture. This sudden change in the received data causes drone to do glitch like behaviour or even crash.

I have tried other output rates such as 250, 500, D500, F500 and result is the same also faster update rate causes failsafe telemetry loss. I had no issues when i was using versions around 2.1.0

I am a drone specialist and i get the screenshots from my flight controller software parallel to betaflight and the below pictures are the usb debug data of raw rc data, flag result with aux switches and the final motor results.

The RC channel order is Roll, Pitch, Yaw, Throttle, Aux1, 2, 3 and so on.

48 is min throttle so disarm value which stops the motors in DSHOT protocol. As you can see motor outputs go crazy after this event.

[image]
[image]
[image]

closed 2022-11-22 by ckflight

AlexKlimaj, 2023-01-13:
Did you figure out what was causing this? I am seeing it on 3.1.2

[image]

ckflight, 2023-01-13:
Hi, it is definitly software related because i did not have any issue when i was using elrs version 2 firmware. I think the link quality in version 3 is bad and it has a failsafe for a moment As a solution on my flight controller software i simply reject the corrupt data. The corrupt data has for example 900 as a aux value which should be 1000 to 2000 so when it send such data the code ignores. And there is 10 15 packets every second like that. The corrupt data has same format and values so you can observ them by ploting received data.