ExpressLRS/ExpressLRS #1969: Perte "connexion" FC Inav F405 wte  avec modul ELRS Naminno 2.4ghz

captflame2121, 2022-11-26:
Bonjour ,
-Ma radio est une Horus x10 avec module Namimno 2.4 gz ( firm 3.01)
- Carte FC Matek F405 WSE ( Inav 5.01) RX ( firm DIY_2400_RX_ESP8285_SX1280-3.0.1 )

Paramètres TX Namimno ELRS /
Packet Rate  = 100 hz
Telem            = on  ou off 
Switch Mode  =  16 ch rate /2

Ma radio est configurée  : 
[image]

Si j'active le chanel  6 ou le chanel 9 sur mon planeur .  La connexion TX - RX se fait toujours ( je le vois dans INAV , les curseurs bougent )   **MAIS PHYSIQUEMENT** il n'y a plus rien , plus aucun  servos ne bougent. Lorsque je bouge les manches sur la radio les servos sont bloqués . Comme si on avait débranché la batterie.

La seule solution est de débrancher la lipo , la transmission TX -RX fonctionne et les servos bougent à nouveaux.

Je n'ose pas aller voler tant que je ne trouve pas le problème car certain que je crash .
Je m'en suis aperçu en faisant des essais à la maison.

Je pense que le problème vient de la transmission ELRS  qui se bloque , pourquoi uniquement sur le chanel 6 ou 9 ?

Si vous avez une solution  ????? merci à vous

captflame2121, 2022-11-26:
Je viens de faire le test sur mon autre drone Rambler RS qui a une carte FC F405-WING SE
Exactement le même problème si j'active le chanel 6 ou le chanel 9 et que je bouge les sticks de la radio  tout se bloque , les servos ne bougent plus mais la connexion TX -TX fonctionne toujours sous INAV.

pkendall64, 2022-11-26:
This is more a question for the discord server in the #help-and-support channel rather than a github issue.
Could you join discord and ask there, in English, please.

captflame2121, 2022-11-26:
Hi , ok can you give me the way please fro put  my problem to help and support .
Thanks 

Bests regards

pkendall64, 2022-11-27:
Go to this website https://discord.com/invite/dS6ReFY and join up, once there you should see a list of channels in the ExpressLRS server, the channel for these questions is the help-and-support channel.

jurgelenas, 2022-11-27:
Closing the issue. @captflame2121 if you had any problems joining our discord server and getting help there please let us know.

closed 2022-11-27 by jurgelenas