ExpressLRS/ExpressLRS #1977: LUA script does not update TX Power setting following change of TX Power

MartinHugh, 2022-12-04:
## Current Behavior
I modified the TX Power from 250mW Dyn to 100mW Dyn.
When I backed out of the edit to the main page, the TX Power still shows 250mW.  
Exiting the LUA and rerunning it shows the correct value.

## Steps to Reproduce
1.  Run the LUA, note the TX Power displayed on the main page

2. Select TX Power to edit it.  

3. Change the TX Power to a different value

4. Back out of the edit page, back to the main page

5. The main page is not updated to reflect the new value. 

## Possible Solution (Not obligatory)
The main page should be refreshed with the set values whenever it is shown.
As it stands it gives the impression that the edit was not completed properly.

## Details
The values set should be the ones which are displayed.

## Your Environment
TX16S
HappyModel ES24TX
ELRS 3.0
Edge TX 2.8

pkendall64, 2022-12-04:
Are you running the updated V3 Lua. Thats one of the things that changed there, when you change a value in a sub-folder the parent folder is refreshed to show the updated value. I've just tested on my TX16S and it's updating immediately, like it should.

MartinHugh, 2022-12-04:
Great shout @pkendall64.  I know I have both on my radio, so I bet I am running the wrong one.  Thanks.

closed 2022-12-04 by MartinHugh