ExpressLRS/ExpressLRS #1984: r9 slim plus OTA

davifelix, 2022-12-08:
Good morning. I have a r9 slim plus ota with serious signal reception problems. From what I remember it never went well. I mean that the link quality reaches zero within 10 meters or so. TBS antennas and drone, firmware elrs 2.51. Do you have any indication 0er try to solve? I also have a regular R9slim Plus that has crazy reception.

jurgelenas, 2022-12-08:
Please try ExpressLRS 3.X first. ExpressLRS is 2.X is no longer supported.

jurgelenas, 2022-12-08:
> Some time ago I read about problems with R9 Mini with Express LRS 3.0. Have they been resolved? or I run on. Continued failsafe risk?

You can flash 3.x-maintenance branch. We are going to release 3.1.1 real soon from it. Just for testing.

jurgelenas, 2022-12-08:
P.S. it has to be flashed to both TX and RX

davifelix, 2022-12-11:
I haven't been able to test yet. tomorrow I see to do something ...... but first I try to see if everything works with original FrSky firmware. I wouldn't want the problem to be hardware......

whinds, 2023-01-02:
Assuming the TBS antennas' connectors you mentioned are ipex1, you should be fine, but its worth double checking. The R9mini's use ipex4 which is different, but the slim+ and slim+ ota use ipex1.

davifelix, 2023-01-02:
I'm perfectly sure of those antennas as they are new ones never used.
anyway...... Even with the latest firmware version I have the same signal
problem. with FrSky firmware no problem. What can I still check before
selling the receiver?

whinds, 2023-01-02:
That’s odd. Maybe check the power on the telemetry setting? Although 10mW
should still be plenty…50mW is default but it could be set at 10mW for some
reason. Does the firmware you selected match the antenna frequency i.e.
915mhz vs 865mhz?

Just for the record this issue is definitely isolated to your own receiver
because I have three different slim+ plus ota and all three have gone out
over a kilometer with perfect signal with the 3.0 and 3.1 firmware.

davifelix, 2023-01-03:
At this point it may be an isolated problem of my recipient. I have at home a sAt this point it may be an isolated problem of my recipient. I have a slim+ that has no problem. If I do not solve I have to sell this receiver unfortunately. 
Thank you all.

closed 2023-04-15 by github-actions[bot]