ExpressLRS/ExpressLRS #1987: BetaFPV nano TX module 3.1.1 Lua Script Frozen

stevewitwitty, 2022-12-09:
## Current Behavior

Updated BetaFPV nano TX 2.4ghz to latest 3.1.1 and lua script gets stuck on loading screen. Same issue when I tried to update to earlier 3.1.0 release.

## Steps to Reproduce

Update 2.4ghz BetaFPV nano tx module to 3.1.1 LBT

## Possible Solution (Not obligatory)

## Your Environment

- TX hardware: BetaFPV Nano TX module 2.4ghz
- RX hardware: Betaflight Nano RX 
- Handset model: X9 lite
- EdgeTX version: 2.8.0
- ExpressLRS version: 3.1.1

stevewitwitty, 2022-12-09:
@pkendall64 I changed the CRSF Baudrate to 400k and that worked! LUA working, thanks. What Baudrate should it be set to out of interrest??

closed 2022-12-09 by stevewitwitty

pkendall64, 2022-12-09:
> @pkendall64 I changed the CRSF Baudrate to 400k and that worked! LUA working, thanks. What Baudrate should it be set to out of interest??

Any baudrate that works for you. Some radios (especially Frsky ones) have slow inverters which means a lower baud-rate must be selected. So basically start at 400k and then try 921, 1.87 etc till it stops and use the last one. The only thing is that some packet rates are not supported on lower baud-rates, but the Lua script will tell you if you select them i.e. F1000 will not work on 400k.

stevewitwitty, 2022-12-09:
@pkendall64 thanks for your help, much appreciated