ExpressLRS/ExpressLRS #1988: Connection lost and recover loop with PWM receivers when flashed with 3.1.1 EU domain LBT

finkalexander, 2022-12-09:
## Current Behavior

Hi, I have an issue with ELRS PWM receivers: BetaFPV ELRS Micro Receiver and Matek ELRS-R24-P6. If I flash them with firmware 3.1.1 EU domain LBT, I got a loop where the conntection works for a few seconds, and then it is completely lost (servos stop working). This repeats. During the connection lost, the RX LEDs signal a connection lost.

## Steps to Reproduce

1. Flash the radio with ELRS 3.1.1 EU LBT

2. Flash the PWM receiver with ELRS 3.1.1 EU LBT

3. Connet radio and receiver with bindphrase, 100Hz Full, 8Ch, Model Match Off

4. Wait for a few seconds

## Possible Solution (Not obligatory)

## Details

Behavior occures with both PWM receivers, and both radios. There is no difference in behaviour.

With my other non PWM receivers (Matek ELRS-R24-V & ELRS-R24-D), also flashed with 3.1.1 EU LBT, the connection works fine. I did not change the settings in the radio. I tested it with a Radiomaster TX16S Mk II ELRS and a Radiomaster Zorro ELRS both flashed with 3.1.1 EU domain. When I flash the receivers with the ISM domain firmware, the connection works fine (TXs still with 3.1.1 EU LBT).

## Your Environment

- TX hardware:: Radiomaster TX16S and Zorro with internal ELRS module flashed with 3.1.1 EU LBT
- RX hardware: BetaFPV ELRS Micro Receiver and Matek ELRS-R24-P6
- EdgeTX version: 2.8
- ExpressLRS: 3.1.1 EU LBT
- Packet Rate: 100Hz Full, 8Ch
- Telemetry Ratio: Std (1:32)
- TX Power: 100mW Dyn

UserDefines for the RX:
-DRegulatory_Domain_EU_CE_2400
-DMY_BINDING_PHRASE="MyBindingPhrase"
-DLOCK_ON_FIRST_CONNECTION
-DAUTO_WIFI_ON_INTERVAL="60"
-DFAST_SYNC

scooper1, 2022-12-09:
I had an issue with sbus converter doing the same thing   I had to bring down the packet rate - diag for the condition change telemetry rate and if the response of the servo is quick or slower then this is the issue -

finkalexander, 2022-12-10:
Thanks for the suggestion. I vary the packet rate and also the telemetry ratio. It seems like there is no effect on the issue, the connection lost, revocer lool is the same. I updated the details above.

paulkidd68, 2022-12-10:
I’m having the same issue. 
I went back to using the 3.x.x maintenance firmware.

finkalexander, 2022-12-10:
You can flash the 3.1.1 ISM version, this will work also.

MartinHugh, 2022-12-11:
I have run into issues trying to upgrade a number of aircraft from 3.0.0 to 3.1.1 LBT
The EP1 RXs all upgraded without issue.

However I have about six aircraft on Matek P24 P6 RXs.
I have tried to upgrade 3 of them (DIY_2400_RX_PWMPEX-3.1.1 (LBT)) 
and they all have the same problem, cycling, dropping the link after initial bind.

If I try to flash again (over WiFi) I get a "not enough space" error.

I then try flashing over UART/FTDI and I get an error saying ELRS cannot see the COM port.
The port is correct, the FTDI is wired correctly, the bootloader button is pressed prior to power up

I am pretty much stuffed at this point, so try to load over FTDI using ESPHome-Flasher.
This at first complains about the COM port, but then switches to do an erase, and then successfully flashes the ELRS bin

So now I can bind OK, not cycling saying 'telemetry lost'
But I have a Model mistmatch, and the PWM failsafe settings are lost

I can now hook back up to the RX over Wifi and so set the model number and the PWM settings.
When I try to bind again, I get cycling, 'telemetry lost' again.

Almost as if resetting the model and pwm settings changed something which affects the ability to maintain the link.

As I said I have 3/3 failures to upgrade this particular RX type. And another 3 waiting to upgrade to align with my TX which is on 3.1.1 LBT.

I am hoping I am missing something really obvious, but if not I will need to roll everything back to 3.0.0 which was solid

My gear is as follows

Radio : TX16S 
TX Module : HappyModel TX24 on LBT 3.1.1.
RX :Matek P24-P6

Flashing DIY DIY_2400_RX_PWMPEX-3.1.1 (LBT)

Any suggestions please.

MartinHugh, 2022-12-12:
OK. Here is a long shot.
I am pretty sure that when compiling firmware for my upgrade from 3.0.0 ISM to 3.1.1. LBT I have had the LBT option selected, and in fact, it is sticky once selected anyway.

But I have wondered why the final binary produced does not differentiate in its filename between ISM and LBT, that could lead to confusion.

So I have LBT 3.1.1. on some PWM RXs which when connected to an LBT 3.1.1 TX, cycle the connection, as per the topic of this issue.

I have a second TX module which has 3.0.0 ISM on it.
When I attempt to connect from that to the LBT RXs it works fine.  No cycling.

So the question is, how can we verify that the configuration management is producing LBT builds, rather than ISM builds ?
I checked a build log, and neither is mentioned.

finkalexander, 2022-12-12:
I can confirm that the settings of the RX incl. the modelmatch ID, channel and  PWM, are random numbers after the upgrade from 3.0.1 LBT to 3.1.1 LBT. I have to reset the setting, and set the model match ID in the web UI.

The model match ID seems to do not have any efect to the connection lost/recover cycle. I tested it with and without model match ID.

In your post before you mentioned, that with a non LBT TX and a LBT RX the connection is stable. For me the other way around works as well: LBT TX with non LBT RX works fine (all 3.1.1).

MartinHugh, 2022-12-12:
> The model match ID seems to do not have any efect to the connection lost/recover cycle. I tested it with and without model match ID.
>
> In your post before you mentioned, that with a non LBT TX and a [...]

Interesting about model match.  I have repeated this, and only ran into it in the first place because I use model numbers and non standard PWM failsafes.  Before I changed the model match the link was fine,

Also it would be nice to know if a hybrid solution would work in the medium term, until this issue is resolved.   It would suit me better to move the PWM aircraft over to ISM and keep everything else including the TX on LBT.  But it is not known if this is supported.

finkalexander, 2022-12-12:
I only tested the hybrid of LBT TX and ISM RX for a few minutes, so no garanty that it will work safely. But would be good to know, because this would be a solution for me as well.

MartinHugh, 2022-12-12:
More anecdotal evidence.  I just flashed a Matek P24 R6 to ISM 3.1.1
It bound fine with LBT TX, but model match was wrong.
I booted to Wifi, changed model match and PWM failsafe settings.
Saved, rebooted.
On power up, the RX LED flashed once and then goes off.

Just repeated this with ISM 3.0.0
Flashed firmware.
Adjusted model match and PWM failsafes from wifi link,
I now have a solid link from LBT TX to ISM 3.0.0.

Something is not right for this RX, it might not be an LBT issue, it might be a 3.1.1. issue as I am having problems flashing and configuring either flavour of 3.1.1

hz89377361, 2022-12-13:
ism 2400 have same problem，diy 2.4 tx with betafpv pwm rx

MartinHugh, 2022-12-13:
> ism 2400 have same problem，diy 2.4 tx with betafpv pwm rx

Can you confirm that you have ISM at both ends,  or a mix of ISM and  LBT ?

MartinHugh, 2022-12-14:
I reported above that this problem did not show itself after I flashed to 3.1.1 LBT until I had modified model match number or pwm settings.  This is true, and was experienced multiple times on at least 3 different Matek P24-P6 receivers.

However, I have now tried to flash a 4th Matek P24-P6, and immediately that I flashed from 3.0.0 ISM to 3.1.1. LBT the cycling connection was evident.

I tried to downgrade back to ISM but the ELRS Configurator reported 'not enough space' for some reason.

I then used ESPHome-flasher to do the downgrade and this worked fine, and the link is behaving itself, but I am technically illegal, as the RX is transmitting as LBT.

hz89377361, 2022-12-14:
> > ism 2400 有同样的问题，diy 2.4 tx with betafpv pwm rx
> 
> 你能确认你的两端都有 ISM，还是 ISM 和 LBT 的混合？ 能否请您说明每一端的固件版本。 谢谢

yes，tx and rx both are ism，but tx tx is flashed from lbt to ism

finkalexander, 2022-12-14:
I tried several configuration, and reflashed my devices (TXs & RXs). Also I reset all devices to their defaults (both runtime and PWM settings). I still get the issue when both TX and PWM RX are flashed to 3.1.1 EU (LBT).

The only way for me to get a stable connection is to keep the TX on 3.1.1 EU (LBT), but flash the RX to 3.1.1 ISM. As mentioned in the release notes for the prev. version.

I have tested it with the BetaFVP PWM RX and the Matek-P24-P6, both RXs behave the same: Cycle between connection los and reconnection. Not only the telemetry is lost, it is the complete connection. The RXs start blinking when the connection is lost.

MartinHugh, 2022-12-14:
> I have been through all these procedures and cannot get this to fail. I only have a BETAFPV PWM RX though. I've tried both TX and RX on LBT, one on LBT the other on ISM and both on ISM. I've set the model match, failsafe by Lua and by wifi and every time it just works as it should.
> 
> BTW the only different in LBT and ISM firmwares are OTA compatible, because the only difference is that LBT may skip sending a packet it it thinks theres something already using the frequency it's on.

Also,  I note that once the problem is induced, the fault is present for all these TX firmwares (ISM 3.0.0, ISM 3.1.1, LBT 3.0.0, LBT 3.1.1) which inconclusively suggests the problem is at the RX end.

pkendall64, 2022-12-14:
> I have tested it with the BetaFVP PWM RX and the Matek-P24-P6, both RXs behave the same: Cycle between connection los and reconnection. Not only the telemetry is lost, it is the complete connection. The RXs start blinking when the connection is lost.

Now this last part is very telling, what you are seeing is the RX reboot. So theres a software or hardware problem that I just can't seem to reproduce. One of the developers had a similar problem but others could not reproduce the problem (once all the PR's were merged). If you connect a serial adapter to the RX/TX pins and set the baud rate to 74880 you will see (if it is actually rebooting) a log message printed out. If you could capture that and post it here that would be helpful.

MartinHugh, 2022-12-14:
I had a go at this @pkendall64 , but my first obstacle is that my Silicon Labs FTDI adapter does not support that baud rate :-(

MartinHugh, 2022-12-14:
OK I changed the Baud rate in putty to 115200, and now see this dumped on each reconnect

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0007a780
~ld

pkendall64, 2022-12-14:
Right, well that settles it. It's a reboot. Now that can happen because of a few things, a software issue or a power supply issue. I'll continue to look at the software side of things.
Perhaps you could disconnect all servos and try with a different power supply and see if it continues to happen that would help narrow it down.

MartinHugh, 2022-12-14:
Originally this was happening with RXs fitted to aircraft, powered by Lipo via a BEC.  
The above ESP debug was done with the RX connected only to an FTDI which was plugged into a USB port on PC.  

Additionally I have just repeated the test with the RX powered on the desktop with a Lipo and a different BEC, and also tested with a bench power supply connected to the RX.

All four cases behave the same.  Reboots every 20-30s or so.  "telemetry lost" on the TX, followed by the LED flashing on the RX, and it reconnects.  Repeat forever.

finkalexander, 2022-12-15:
> Now this last part is very telling, what you are seeing is the RX reboot. So theres a software or hardware problem that I just can't seem to reproduce. One of the developers had a similar problem [...]

I can confirm the serial output, it is the same for both the BetaFPV Micro PWM and the Matek-R24-P6 both with 3.1.1 EU (LBT):

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0007a780
~ld

If tested/seen this behaviour with different power supplys FTDI adapter, but also a 5A BEC powered by a full LiPo.

MartinHugh, 2022-12-15:
> Right, well that settles it. It's a reboot. Now that can happen because of a few things, a software issue or a power supply issue. I'll continue to look at the software side of things. Perhaps you [...]

Presumably code sections have to be subscribed to the watchdog(s) in order for a watchdog interrupt to fire.  Does this therefore identify a set of candidate code sections which could be causing this issue ?

pkendall64, 2022-12-16:
Could I get users having this problem to test PR #1998 to see if it fixes the problem for you.

finkalexander, 2022-12-16:
I've flashed a Matek-R24-P6 and BetaFPV Micro PWM with the Potential fix for PWM reboot #1998. Even after a few minutes I didn't got any connection lost & recover loop (reboot).

If tested the fix with a Zorro flashed with official ELRS 3.1.1 EU (LBT) and both receivers mentioned above flashed with #1998 EU (LBT). I also tested different packet rates.

MartinHugh, 2022-12-16:
See comment 
[https://github.com/ExpressLRS/ExpressLRS/pull/1998#issuecomment-1354031777]

Paulraaf, 2022-12-16:
> Could I get users having this problem to test PR #1998 to see if it fixes the problem for you.

This fixes the issue for me. Thanks!

DocTrucker, 2022-12-19:
I am having the same issue with a JMU PWM rx which is fine on 3.0.0. I will try the suggested fix tomorrow and report with a full set of settings.

Thanks.

PR #1998 (2022-12-17): Potential fix for PWM reboot
We found a couple of things that needed cleaning up in the serial code, which the PWM should not be using but was creating a Serial port object that seems to be causing the problem with the reboot.

I (finally) managed to reproduce the issue described in #1988, and these changes have resolved the problem for me.

Fixes #1988
closed 2022-12-27 by pkendall64