ExpressLRS/ExpressLRS #1989: QX7 Lua 3.1.1 betafpv micro tx

Kasg04, 2022-12-10:
## Current Behavior
lua script is freeze on loading

When tring to run lua script v3 its freeze on loading
## Steps to Reproduce
After upgrade betafpv micro tx to 3.1.1 , i cant run lua script v3

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

stevewitwitty, 2022-12-10:
I had a problem similar to this, I had to change the baud rate to 400k. Originally it was set to 115k and lua script froze. After I changed the baudrate to 400k it worked fine. Maybe worth a try

Kasg04, 2022-12-10:
I reduce the refresh rate to 100mhz and every thing is working right now

Kasg04, 2022-12-11:
> Joshua Bardwell has a video covering errors like this. See link and go to approx 10.30 on video: [Bardwell video express LRS](https://youtu.be/cyILQqxPpZI)

Thamk You!

closed 2022-12-11 by Kasg04