ExpressLRS/ExpressLRS #1992: RSSI of one radio is lagged on dual-radio Rxs

David-OConnor, 2022-12-11:
## Current Behavior

Using 3.1.1 (Notable due to reworked dual-radio RSSI behavior; bug is reproducable on prev versions, but I'm sticking with 3.1.1's behavior in this description. One of the radios' RSSI lags, as reported in the link stats packet. Ie, on single-rad Rxes, and on the second radio on dual-rad, response to, let's say, plugging in or unplugging an antenna is very fast. On rad1, it takes on the order of 30-60s to settle. I've reproduced this on the AnyLeaf dual-radio Rx, and the Happymodel EP1 dual.

## Steps to Reproduce

- Connect a dual-radio ELRS Rx to a FC capable of display link stats; pair it to a Tx.
- Observe RSSI1 and RSSI2 with no antennas connected.
- Plug an antenna into radio 2. Observe RSSI2 change, and time-to-settle.
- Unplug antenna. Observe  RSSI2 change, and time-to-settle.
- Plug an antenna into radio 1. Observe RSSI1 change, and time-to-settle.
- Unplug antenna. Observe RSSI1 change, and time-to-settle.

## Details

Of note, In the WiFi config under `hardware.html`, I tried swapping all radio1 and radio 2 pins, so as the radio previously denoted radio 1 was radio 2 and vice-versa. With default pins, RSSI1 is the lagged one. After swapping pins, RSSI2 is the lagged one. This is interesting, as the same physical radio is the lagged one in both cases.

## Your Environment

- TX hardware: TX16S + happymodel Tx
- RX hardware: AnyLeaf dual-radio ELRS receiver, or Happymodel EP1-dual/TCXO
- Handset model: (Not sure what this is?)
- OpenTX version (including nightly number) (Not sure what this is?)
- ExpressLRS version (TX & RX MUST MATCH): 3.1.1
- Packet Rate: Tested at 500Hz LoRa
- Telemetry Ratio: Default
- user_defines: (?)

blind-oracle, 2022-12-22:
In fact dual-radio RXes work in ELRS more or less the same way as dual-antenna RXes.

Because of this the active radio is only changed when we either have a shard drop of RSSI or a packet loss.
So only the RSSI of active radio is updated, the other one stays constant until your RSSI drops by 5dbm or a packet is lost -> then the radios are switched and the other one becomes active.

David-OConnor, 2022-12-22:
Shouldn't it still be able to give you 2 independent RSSIs, since there are 2 independent RF pipelines? (That feed to the common MCU that can mix/match data A/R)

blind-oracle, 2022-12-22:
> Shouldn't it still be able to measure and display 2 independent RSSIs, since there are 2 independent RF pipelines? (That feed to the common MCU that can mix/match data A/R)

The way it's currently implemented - no, it only updates the RSSI of currently selected antenna (or radio in case of dual RXs). Maybe in the future developers will improve that, but currently it works like this.

> Could you elaborate on burning our RX radios without antenna? What's the mechanism on that? Thank you.

ExpressLRS RX is not much different from TX, it sends the telemetry back, EP1 has 19dbm power and AnyLeaf 22dbm. If this energy has no way to radiate normally through an antenna - it might overload the amplifier and it'll blow.

David-OConnor, 2022-12-22:
I'm happy to close this issue as working-as-intended. And if anyone else notices the same, they'll find this in the search. (And perhaps the behavior will change in the future)

closed 2022-12-22 by David-OConnor