ExpressLRS/ExpressLRS #2010: 3.1.1 Lua is non responsive and non functional

Thoemse, 2022-12-20:
When flashing 3.1.1 the lua takes about 10 minutes to load. This can be helped a tiny bit by increasing the bauf rate. Once loaded the options are not working at all. Reverting back to 3.0.1 fixes the problem. I removed the lua and luac files and replaced with the new ones from Configurator.

Hardware used: BetaFPV 2.4Ghz Micro with a Flysky NV14

Thoemse, 2022-12-23:
Same problem with 3.1.2. Going back to 3.0.1 solves it.

Oxnitron, 2022-12-30:
Hello,
 I have the same issue. LUA is not loading on 3.1.2 with Betafpv micro 1W module :(

nerdCopter, 2022-12-31:
same, with:
latest `elrsV3.lua`.
BetaFPV Nano TX 2.4Ghz.
Edge TX 2.8.0 nightly.
frSky xLite original.

`Loading...`
[image]

works fine on my R9M Lite module w/ ELRS v2.5.1, however.

Oxnitron, 2022-12-31:
I found the solution for me.
In the radio settings (external rf) you have to change the baudrate to a higher value than 125.
Putting it on 400 fixed it for me.

Hope this helps you too.

[2022-12-31_15-30]<https://user-images.githubusercontent.com/56646290/210155967-4d90f342-9f8e-4689-b853-da550dfe0187.png>

nerdCopter, 2022-12-31:
> [...] Putting it on 400 fixed it for me. [...]

Solved! TYVM!!

Thoemse, 2023-01-02:
Doesn't solve it for me. 
Upping the baudrate makes it load within minutes instead of not at all but it stays unresponsive and unusable.
Beta FPV Micro 500W version on a Flysky NV14

urlu75, 2023-01-02:
Same for me. Changing the baud rate does not solve the issue, still unresponsive and unusable (and does not work in getting the ardu param).

2xWhiskey, 2023-01-31:
Switching to 400 baudrate on my TX16S got my elrs3 LUA working.  Thank you for the solution.

ptatrai, 2023-02-01:
I have the same issue. HM TX PRO 2.4GHz with TX16S. Lua v3 script can be loaded when a higher bitrate is set, but I cannot change any option. After pressing enter, options revert back to the original values. It works fine with 3.0.1 or 2.5.1. Doesn't work with anything newer (up to 3.2.0).

dcseek, 2023-02-04:
> Today I tried to figure out which commit broke it by git bisect. I was building FW from a local repo clone. I wasn't able to reproduce the issue at all and somehow it fixed for me also the normal build of the official releases greater than 3.0.1. So currently I'm not able to reproduce the issue at all.

Hi I would like to know wich repo you used I have the same issue with nv14

psfales, 2023-02-16:
I think I'm seeing the same thing here.  I have a Taranis QX7 running OpenTX 2.3.12.    Expresslrs 3.0.0 and 3.0.1 work fine but 3.1.1 and 3.2.0  hang when I run the lua script (which seems to be the problem the OP had), and my screen looks like the photo in the comment from 12/31/22.

psfales, 2023-03-07:
As @Thoemse pointed out, it seems like it must be software related because going back to 3.0.1 fixes the problem.  Any ideas where to look in the code?   I'm willing to poke around a bit, but since I know nothing about the codebase, it's hard to know where to even start.

pkendall64, 2023-03-07:
I imagine that it would be because we've added more to the Lua parameters returned to EdgeTX between those two releases.

psfales, 2023-03-07:
Would that explain why changing the baud rate from 115K to 400k fixes the problem?

dcseek, 2023-03-07:
> I imagine that it would be because we've added more to the Lua parameters returned to EdgeTX between those two releases.

Thanks, would be nice to understand what was added and if it is mandatory in v3 to work. Maybe we could find a workaround. For now nv14 and qx7 radios are stucked with 3.0.1....

Thoemse, 2023-03-07:
> Would that explain why changing the baud rate from 115K to 400k fixes the problem?

It doesn't.

psfales, 2023-03-10:
As another data point, I was able to try my module (HM TX PRO 2.4GHz) in a Radiomaster TX12.   It is able to run at 400K  (does not need the inverter mod), and I was able to use it to install expressLRS 3.2 and connect to it with the lua script.   However, when I set the baud rate to 115K, it hung in the manner described by the OP.   I could get it working by switching the baud rate back to 400K OR by reinstalling 3.0.1 and running at 115K. 

This makes me think that anyone with 3.2 installed could reproduce the problem by setting the baud rate to 115K on (I assume) any radio.  I don't know if the expressLRS hardware matters, but folks seem to be reporting it for several different platforms.

nurikk, 2023-03-15:
Same issue here, TX16S internal module. Changing baud rate to 400 didn't work for me. Everything works only if I rollback to 2.5.2. I flash firmware using edge tx passthrough

ewasscher, 2023-03-22:
Same issue here with a HM ES24TX Pro and Taranis X9D+ and TX firmware > 3.0.1. Setting the baud rate higher that 115200 isn't an option. That just doesn't work, probably because of a slow inverter circuit as mentioned here: https://www.expresslrs.org/hardware/x9d-troubleshooting/

chocotov1, 2023-03-25:
> I think I'm seeing the same thing here. I have a Taranis QX7 running OpenTX 2.3.12. The module is HM TX PRO 2.4GHz.
> 
> Expresslrs 3.0.0 and 3.0.1 work fine but 3.1.1 and 3.2.0 hang when I run the lua script (which seems to be the problem the OP had), and my screen looks like the photo in the comment from 12/31/22.
> 
> My QX7 is limited to 115k because I'm not yet willing to install the Crossfire mod (https://blog.seidel-philipp.de/fixed-inverter-mod-for-tbs-crossfire-and-frsky-qx7/) so switching to 400K is not really an option.

@psfales I'm running the exact same setup and have the same problem.
- QX7 (no mods)
- HM TX PRO 2.4GHz
- 115K

With 3.2 loaded onto the module the lua script says "Loading..." and nothing happens. With 3.0 and 3.0.1 the lua script works. Not sure how it flies though, as i'm still waiting to receive my first elrs quadcopter.

psfales, 2023-03-25:
Thanks for the additional datapoint.     It seems pretty consistent that users of transmitters with the inverter problem (QX7, nv14, and ??)  are stuck on 3.0.1 and can't go to 3.1 or 3.2.

It would be helpful if someone folks that have ExpressLRS 3.1 or 3.2 working could try testing at the 115K baud rate and report their results.   If (as I suspect) the lua script breaks for anyone running at 115K, it might make it easier for developers to reproduce and debug the problem.

nurikk, 2023-04-12:
Hello everyone, I had this issue with Lua script stuck in `Loading....` after updating from `2.5.2` to any `3.x.x` using edgetx passthrough.

Solution helped to me: Fist I've updated to `3.0.0` **using WiFi** page (enable wifi in Lua menu, connect to wifi and navigate to http://10.0.0.1

psfales, 2023-04-16:
Just checked 3.2.1...  still not fixed.

psfales, 2023-04-16:
> Hello everyone, I had this issue with Lua script stuck in `Loading....` after updating from `2.5.2` to any `3.x.x` using edgetx passthrough.
> 
> Solution helped to me: Fist I've updated to `3.0.0` **using WiFi** page (enable wifi in Lua menu, connect to wifi and navigate to http://10.0.0.1

The pattern most folks seem to be seeing is that 3.0.1 and older works (including 3.0.0 and 2.5.2) while anything newer  (3.1.x and 3.2.x)  has the problem.  This seems to be another data point confirming that pattern.       The key factor is not that WiFI is used for the update, but that the version is 3.0.0.

pkendall64, 2023-04-16:
I just tested this with 3.2 on a TX16S at 115k baud and it happily loads the Lua script, albeit slowly.
Have you tried setting the "sample mode" option in EdgeTX to "One bit" that may help and may even allow you to select 400k.

psfales, 2023-04-16:
> I just tested this with 3.2 on a TX16S at 115k baud and it happily loads the Lua script, albeit slowly. Have you tried setting the "sample mode" option in EdgeTX to "One bit" that may help and may even allow you to select 400k.

Thanks for the data point.  That implies that there may be more than one factor in play (which, sadly, may make it harder to reproduce).    I just re-ran the test on a TX12 with 3.2.1 and got the same results.   (Works at 400k, but gets stuck "Loading"  at 115K. 

When you say "albeit slowly," how slow is that?  I waited at least a minute, but saw no sign of anything happening in that time.

pkendall64, 2023-04-16:
> When you say "albeit slowly," how slow is that? I waited at least a minute, but saw no sign of anything happening in that time.

Sorry, my test was wrong. With an external module at 115k baud it is not loading. That at least gives me something to go on.

psfales, 2023-04-16:
I think that's good news!    It's consistent with my theory that anyone can reproduce the problem by setting the baud rate to 115K, even when their hardware is able to run at 400K

pkendall64, 2023-04-16:
Another interesting thing to note. If I put my module at 150Hz packet rate it loads at 115k baud. Any faster and it fails.

psfales, 2023-04-16:
> Have you tried setting the "sample mode" option in EdgeTX to "One bit" that may help and may even allow you to select 400k.

I'm aware the using one-bit sample mode in EdgeTX is another way to address the problem.   But some folks (like me) are using OpenTX where that's not an option.

psfales, 2023-04-16:
> Another interesting thing to note. If I put my module at 150Hz packet rate it loads at 115k baud. Any faster and it fails.

Yes, that's interesting.    It could be important information for anyone attempting to reproduce the problem.    

I know nothing about that code, but that makes we wonder if 150Hz is the fastest packet rate that can be sent at 115K and if attempting to send packets faster than that results in an overrun and some of them getting lost.    (You mentioned on March 6 that more lua parameters are being returned - does that require more packets to be sent?)

psfales, 2023-04-16:
> Another interesting thing to note. If I put my module at 150Hz packet rate it loads at 115k baud. Any faster and it fails.

Reproduced the same thing here on my QX7.    With the 150Hz packet rate, it works at 115K!

psfales, 2023-04-21:
Do we know what changed in 3.1.0 that reduced the maximum rate to 150Hz?    (It's just curiosity -  I doubt anyone is going to change anything, but it would nice to know why a purely software change had that effect!)

pkendall64, 2023-04-21:
> Do we know what changed in 3.1.0 that reduced the maximum rate to 150Hz? (It's just curiosity - I doubt anyone is going to change anything, but it would nice to know why a purely software change had that effect!)

I went thorough all the code changes and reverted them locally but that didn't fix it. There was two other things that changed.
1. upgrade to the base platform from 3.4.0 to 5.1.1 and my suspicion is that the problem lies there. I just have not had time to test that theory.
2. shift processing of the CRSF from core 0 (the secondary core) to core 1 (the main loop core). This was done so all main processing code in the main loop code and all secondary devices are handled in the second core and the second core can sleep for longer periods to reduce power consumption.

TL;DR even if we found the problem I don't think we'd be reverting any of those changes as they are more beneficial to the project.

nerdCopter, 2023-04-21:
the 5.1.1 reference for any onlookers https://github.com/ExpressLRS/ExpressLRS/pull/1831/files

psfales, 2023-04-21:
Thanks for looking into it.

closed 2023-04-25 by pkendall64

psfales, 2023-07-29:
Just updated my TX from  ELRS v3.2.1 to v3.3.0 and it broke the Lua script.   (Hangs during "Loading...") 

I was able to get it working by reverting to 3.2.1 (so I'm pretty sure that 3.3.0 is the issue).  But then I found that I could get it working by setting the baud rate on the QX7 to 921K (instead of 400k).   I wasn't aware until now that setting it higher than 400k was even an option, and I also wasn't sure whether the inverter mod would run at a higher speed than 400k, but apparently it does. 

So, posting here for two reasons:
1) Maybe it will help someone else who is running into  the same problem
2) Trying to find out if this is known/expected behavior.   If it causes breakage for anyone using 400K or slower, I'm surprised that it's not mentioned in the release notes, and that there don't seem to be any bug reports.

nerdCopter, 2023-08-17:
> > [...] Putting it on 400 fixed it for me. [...]
> 
> Solved! TYVM!!

well it was solved until more recent firmware updates. (or maybe recent edgetx updates :man_shrugging: )
i'm now required to go from 400k to next level up (whatever it is).
what's the bloat that is causing this?

pkendall64, 2023-08-17:
> well it was solved until more recent firmware updates. (or maybe recent edgetx updates 🤷‍♂️ ) i'm now required to go from 400k to next level up (whatever it is). what's the bloat that is causing this?
> 
> side note: BetaFPV Nano TX was/still less responsive than FrSky R9M Lite. (i.e. BFPV-nano seems to require higher baud than R9M-Lite)

We are not 100% sure what is causing the need to upgrade the baud rate, but I am still inestigating.
For your side note, the BetaFPV will be slower to load than the R9 because it has a backpack so it has a lot more Lua entries to load from the module.

psfales, 2023-08-20:
> well it was solved until more recent firmware updates. (or maybe recent edgetx updates 🤷‍♂️ ) i'm now required to go from 400k to next level up (whatever it is). what's the bloat that is causing this?

I'm thinking it's the elrs version.   I'm on OpenTX (so no opentx updates for a long time),  but when I went from elrs 3.2.1 to 3.3.0 I had to change by my baud rate from 400k to 921k (the next level up)

psfales, 2023-11-06:
I see that ELRS 3.3.1 is out and the release notes say that  "Users with radios that are limited to 400kbaud and Lua freezing at the loading screen" should upgrade from 3.3.0 to 3.3.1

PR #2170 (2023-04-25): Max rate of external module on 115k baud is 150Hz
From 3.1.0 115k baud external modules no longer support 250Hz, so we've reduced the max air rate for them to 150Hz.
Also fix signed-ness calculations for OpenTX sync offset.

Fixes #2010