ExpressLRS/ExpressLRS #2014: Can not update RX from 3.1.1 to 3.1.2 via WiFi - TARGET MISMATCH

bigbear2001, 2022-12-23:
When updating RX from 3.1.1 to 3.1.2 via wifi have error "TARGET MISMATCH"
Force Flash does not help
Reverting to 3.0.1 fix this issue. 

I tested different combinations

unsuccessful attempts
 3.1.1 -> 3.1.2
 3.1.1 -> 3.1,1
 3.1.2 -> 3.1,1
 3.1.2 -> 3.1.2

successfully attempts
 3.1.1 -> 3.0.1
 3.1.2 -> 3.0.1
 3.0.1 -> 3.1.1
 3.0.1 -> 3.1.2

Used hardware:

HappyModel EP 2400 RX
JheMCU SP24S as DIY_2400_RX_ESP8285_SX1280
DIY_900_RX_ESP8285_SX127x

ExpressLRS Configurator v1.5.4
ExpressLRS Configurator v1.5.0
[ExpressLRSBuildLog_20221223210637041.txt](https://github.com/ExpressLRS/ExpressLRS/files/10297434/ExpressLRSBuildLog_20221223210637041.txt)
[image]

Denethor85, 2022-12-24:
Same problem with BetaFPV 2400 Lite RX

VendettaPrime, 2022-12-26:
same problem 3.1.1 -> 3.1.2 HappyModel EP1

pkendall64, 2022-12-27:
We did some investigation today in regards to this issue and the problem is that the new firmware is quite a bit larger with all the features added and is now bigger than the free space in the flash. Luckily the ESP8285 allows uploading a gzipped firmware so we have a PR (#2019) for flashing compressed firmware which should solve this going forward.

bigbear2001, 2022-12-27:
> free space in the flash

but...
When flash 3.12 via UART/BetaflightPassthrough done it is fine.
Problem **only** when flash from 3.11 to 3.12 via wifi.

pkendall64, 2022-12-27:
> > free space in the flash
> 
> but... When flash 3.12 via UART/BetaflightPassthrough done it is fine. Problem **only** when flash from 3.11 to 3.12 via wifi.

Yes, thats because via wifi it stores the new firmware in the left-over space and on reboot it is installed in the right place.
But via UART/BetaflightPassthrough it overwrites the old one directly.

dannyact, 2022-12-27:
Same problem over here: After flashing back the receiver to V.2.5.1 with the FTDI UART method the update via WiFi to V3.1.2 works. However, after testing the V3.1.2. update and connecting to WiFi again for updating, the ELRS site shows "TARGETS MISMATCHING" again ! My guess: something must be wrong in the latest V3 firmware for connecting to ELRS site ? Connecting via WiFi with the V2.5.1 version works excellent. Not so with V3 firmware.

PR #2019 (2022-12-28): Build and flash gzipped binary on ESP8285
When performing wifi build/flash on ESP8285 based devices, we now gzip the firmware and flash that instead as theres not a lot of free space in the flash with the latest firmwares.
fixes #2014
closed 2022-12-28 by pkendall64