ExpressLRS/ExpressLRS #2027: elrs 3.* seems to use more power than 2.*

wang322588, 2023-01-03:
## Current Behavior
I use t-lite and DIY eps32 1280 F27 target.While I updaet to version elrs 3.* I found my tlite battery die quickly.

Maybe it is mot a problem. sx1280 power code may change the powerlevel. But I perfer efficiency :),
## Steps to Reproduce
Flash 3.0.0 and higher

## Possible Solution (Not obligatory)

I did a test.I used a DC power with output power display to my t-lite .Setup 10mw (dyn off) is 1.2W in 2.*  compare 2W in 3.*
## Details

If use less power with stable connection maybe bette的，especially a 1s TX!

## Your Environment

- TX hardware:jumper tlite
- RX hardware: none
- Handset model:
- OpenTX version (including nightly number) edge tx
- ExpressLRS version (TX & RX MUST MATCH): 2.0 and 3.0
- Packet Rate: 150hz
- Telemetry Ratio:1:32
- user_defines:

nightflasher, 2023-01-29:
I can confirm that!
I've tested yesterday the same with my diy E28 tx module (with backpack). Have to charge the Li-Ion more often than before.

pkendall64, 2023-04-01:
We've done some work on reducing the power consumption in the master branch, which will become 3.3 if you'd like to check.

wang322588, 2023-04-03:
@pkendall64 I tested the master build code and 3.2.1 shows the same with 3.2.0. All use more power than version 2.*

CapnBry, 2023-04-06:
Woof. This was tough to get an apples to apples comparison. Namimno Flash OLED on 150Hz, TLM off, 25mW. Powered via XT30 at high enough voltage to draw nothing from handset (also tested freerunning on 3.x, similar power usage)
3.2.0: 181mW @ 9V = 1.629W
2.5.2: 197mW @ 9V = 1.773W
I'd call that the ballpark of same power, or even lower in favor of 3.x. I do not have a E28 DIY to test with. Note that the 2.x value is higher than Sunjun's test on 2.2.0 on 500Hz TLM off = 1.552W at 7.4V input, but it is difficult to compare the numbers since I guarantee our test procedures (and transmit duty cycles) are different.

In any case, there should be no reason that our power usage should be significantly higher on 3.x over 2.x. We'd have to do something amazing to turn 1.2W of power usage into 2.0W and there's just not anywhere we could shove more power. The only things I can think of are:
* Mistake in defining the DIY E28 target somehow blasting the SX1280 power output -- appears to be OK
* PDET code raising the power to max because it shouldn't be used -- does not _appear_ do be doing this?
* ... that's all I can think of. Even if the power amp pins were wrong and we left it on, that would be a pretty minor power usage

I follow our power usage pretty closely and have only seen our usage go down over time. If there's anything, it would have to be specific to this target and not the system as a whole. I do not have this DIY hardware though, so it would be up to you to determine what issue you're experiencing.

closed 2023-05-04 by github-actions[bot]