ExpressLRS/ExpressLRS #2028: EU LBT firmware loosing connetion and doesn't re-connect

mha1, 2023-01-03:
## Current Behavior

LBT EU firmware 3.1.2 at 50Hz packet rate and traditional 2.4Ghz radio close to force LBT activity: After successfully establishing connection the tx module randomly loses connection indicated by module status screen showing “not connected”, rx LED flashing and loss of telemetry and control. After this no further re-connection is established. Power cycling the module establishes connection again.

Notes:
- ISM firmware works as expected. This problem seems to be EU LBT related
- This problem occurs in all 3.x.x versions
- The problem occurs with different receivers which makes the tx module prime suspect
- This problem is more prominent at lower packet rates, most prominent at 50Hz
- This problem occurs more frequently with 50Hz packet rate compared to rarely happening at 100Hz packet rate
- Once connection is lost only power cycling the module will establish a connection again
- Once connection is lost power cycling the receiver doesn’t make a difference - no re-connection
- Once connection is lost the module is still responsive. Changing to a different protocol leads to a “no handset” message, changing back to CRSF shows ELRS status again
- This problem might be partially caused by temperature drift as the probability of loosing the connection declines with the module warming up over time.

## Steps to Reproduce

1. set parameters as per section "Your Environment"
2. have a separate traditional radio close to force LBT activity
3. best to use cold module
4. establish connection
5. wait for the connection to be lost

## Details

see problem description

## Your Environment

- TX hardware: BetaFPV micro 2.4GHz 1000mW
- RX hardware: Matek diversity 2D, Skystars Diversity, BetaFPV lite
- Handset model: Jumper T16, Graupner MZ-32 with CRSF protocol enabled

- OpenTX version (including nightly number): EdgeTX 2.8
- ExpressLRS version (TX & RX MUST MATCH): 3.0.0 up to 3.1.2
- Packet Rate: 50Hz and 100Hz
- Telemetry Ratio: 1:2
- user_defines: TLM_REPORT_INTERVAL_MS 100

mha1, 2023-01-04:
additional note:
- this problem can not be reproduced in a clean rf environment.
- Having a traditional 2.4Ghz radio nearby leads to the described problem and might be a strong indication that something is going on with the ELRS  LBT handling after hitting occupied channels.

MartinHugh, 2023-01-04:
@mha1 In order to help reproduce this, is it worth adding what power settings your two radios were on, and what the distance between TX1/TX2/Rx was when the problem occurred.  I realise you stated 1000mW for ELRS, but not sure whether this is MAX or actual. Can we assume that the "traditional" radio was 100mW ?

Thanks

mha1, 2023-01-04:
Martin - Thanks for looking into this. With traditional radio I mean any EU legal hopping 2.4Ghz radio (100mW, LBT), e.g. Graupner HoTT MZ-32. The ELRS power settings don't really matter. Try 100mW dynamic power. Distance means workshop distance, about 5ft. It's not a problem of radios/teceivers being to close. ISM works fine with the same setup. LBT makes the difference here.

mha1, 2023-01-05:
Further notes:
- With TLM_REPORT_INTERVAL_MS at default 240LU and also turning off dynamic power the problem is still present
- 99.9% sure this is a problem with LBT handling

MartinHugh, 2023-01-05:
@mha1 I am so far unable to reproduce this.   My gear is as follows:

ELRS TX is a TX16S with an internal (Radiomaster) module, running EdgeTX 2.8
Setting are 50hz/Std/Wide/On/Dynamic TX Power

Traditional TX is a QX7 using FrSky D16, running Open TX 2.2.1

I positioned the QX7 in between the ELRS TX and the Reciever (Matek P24 D6) and power it on.
I then powered on the TX and RX.
There was an expected initial delay of 2-3s before they connected.
I left them on the bench for more than 10mins without any problem.

mha1, 2023-01-05:
@MartinHugh Martin, thank you very much for trying to reproduce this. I wouldn't exclude hardware problems but why does ISM work flawless? The root cause may still be hardware related, e.g. parts tolerances together with maybe too tight firmware timing constraints. Can you check the critical timing portions of the LBT code and point me to where to start experimenting?

Oh, and maybe you can get me going on debug outputs: https://github.com/ExpressLRS/ExpressLRS/discussions/2034

MartinHugh, 2023-01-05:
@mha1 there are plenty on here who I am sure can point you at the right place, but although I am a developer, I am not familiar with this section of code.  It is on my TODO list though :-)

mha1, 2023-01-05:
@MartinHugh Well, there's no better learning opportunity than a difficult problem, right 😀?

MartinHugh, 2023-01-05:
> @MartinHugh Well, there's no better learning opportunity than a difficult problem, right 😀?

Will be interested to see what you find 👍

mha1, 2023-01-13:
@MartinHugh 

We did a lot of debugging and found once the problem occurs there are no more frequency hops which ultimately leads to the disconnect and no re-connect. If LBT detects a collision the packet transmission is omitted which we believe messes leaves some status variables in an old or indefinite state which might cause the hopping problem. To test the theory we implemented a lowest power transmission in case of non-clear channel. This worked in so far that there were still disconnects but with re-connects. Only implementing the same strategy for the rx software seems to fix both the disconnect and re-connect problems.

We still think there is some timing issues probably induced by a slightly off crystal. The implementation of lowest power transmissions in case of LBT collisions might just cover a system robustness issue so e believe it is not a valid fix. If you'd try to go for 868MHz LBT it is also no option to transmit for detected collision because of regulatory requirements.

We would very much appreciate you looking into this and will support your efforts by offering to send our module to your address. Please dm me on discord Mickey123#1894

vanurag, 2023-01-31:
We came across this issue recently with a similar setup: (BetaFPV micro 2.4GHz TX + Mateksys R24D diversity RX) on EU LBT setting. We saw this only after having multiple drones with same setup in the near vicinity (within few meters apart in the office). @mha1 were you able to reproduce this issue with a different transmitter HW? I have been testing RadioMaster Ranger Micro TX, but never had the issue so far. Would be good to know if this is specific to BetaFPV TX.

mha1, 2023-01-31:
@vanurag We only have access to BetaFPV micro 2.4G 500mW and 1W modules. Only the 1W module showed the problem. As written in the early description a hardware issue (e.g. temperature drift) might force the problem more dominantly but we are sure you might see this problem in "dirty" environments leading to multiple LBT collisions in sequence. PR #2055 fixes the problem with our 1W module.

closed 2023-04-03 by pkendall64