ExpressLRS/ExpressLRS #2052: Taranis X9D: Lua script execution error: Script panic not enough memory

GramontABAPEnvironment, 2023-01-18:
## Current Behavior

First I installed ExpressLRS lua script on OpenTx 2.3.13 it was working fine.
Then I decided to upgrade the whole radio to newest EdgeTX (2.8.0) along with SD card content update.
Radio is Taranis X9D.

## Steps to Reproduce
expresslrs lua script copied to its place along with other scripts like BF, CRSF lua scripts.

try to execute the script from tools menu, ends up in error: Script panic not enough memory

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: Frsky R9M
- RX hardware: -
- Handset model: Taranis X9D
- OpenTX version (including nightly number) **EdgeTX** (2.8.0) 
- ExpressLRS version (TX & RX MUST MATCH): v3 with v3 lua script
- Packet Rate:
- Telemetry Ratio:
- user_defines:

GramontABAPEnvironment, 2023-01-19:
downgraded to EdgeTX 2.5.0 now it looks like okay...

MaxBearRc, 2023-01-24:
I'm having the same problem too, is it possible to share the custom script if it solved it ?

> I have the same problem with my Taranis X9D+ v2.8.0 flashed with EdgeTX Buddy web flasher. So i created Custom ELRS LUA that has ColorLCD, TouchScreen and Simulator Test code removed.

Ezikyel, 2023-01-27:
> I have the same problem with my Taranis X9D+ v2.8.0 flashed with EdgeTX Buddy web flasher. So i created Custom ELRS LUA that has ColorLCD, TouchScreen and Simulator Test code removed.

Same problem here. Can you kindly send me your custom lua script pls.

paulo-serrao, 2023-01-30:
My workaround: 
*downgraded to EdgeTX 2.7.1
*run ELRS lua once so Taranis could compile the code and saves the .luac into the sd card
*upgraded back to EdgeTX 2.8.0

Runs fine now

Ezikyel, 2023-01-31:
Hey, I'll try. Thank you so much.

snooperace, 2023-02-09:
I am haveing the same issue with a Taranis xd9+ I down graded to Edgetx 2.71 installed the lua script provided by betafpv with their micro tx module firmware and all I am getting is "Loading" when running the Lua script

paulo-serrao, 2023-02-09:
> I am haveing the same issue with a Taranis xd9+ I down graded to Edgetx 2.71 installed the lua script provided by betafpv with their micro tx module firmware and all I am getting is "Loading" when running the Lua script

I would suggest evaluate if you can upgrade to elrs 3.2 (TX and RXs) and use latest lua from ELRS team.

cmanley, 2023-02-10:
Same problem here on my X9D, but I have a question about the downgrade -> upgrade workaround: are the *.luac handset independent, because if so, then I can just copy them off of another handset (the Jumper T-Lite) where they did compile.

pitts-mo, 2023-04-24:
X9D+ with EdgeTX 2.8.x I still can't get the lua to run

first run results in message:
Script syntax error
not enough memory

additional runs result in top bar reporting:
Loading...

Tried the downgrade to EdgeTX 2.7.0 trick and it does not hit error just goes straight to top bar reporting:
Loading...

pitts-mo, 2023-04-24:
Really weird, elrs v3 lua finally started to work on my x9d+.

I started to suspect CRSF was just not working from EdgeTX on my x9d+...  I found once I re-selected CRSF for a model I could use any of my CRSF models. But if I booted EdgeTX directly into a model that did not have CRSF re-selected then modules would not work until after I re-selected CRSF for that model. I also deleted the elrsv3.luac and confirmed it did rebuild without error on 2.8.3.

extra info:
I currently suspect this could have been a quirk related to model migrations. I have both an RM9 and a DIY 2.4 TX for this x9D+ and neither would work with the LUA after updating to EdgeTX so I just had not flown CRSF from my x9D+ since.

Kaffiflight, 2023-06-14:
Similar Problem. 
X9D (classic) here. I am on OpenTX 2.3.14 OneBit. Was able to get rid of repeated telemtetry loss/recoverd reports.
ELRS 3.2.1 on a Radiomaster Micro Module.
ELRS LUA loads half way, then breaks up with the "not enough memory" issue.
Is there a way to get a "smaller" version of the script? I read someone had altered an older version in early ´23 to get rid of the commands for touchscreen.
I am not able to do such thing.

epareja2, 2023-06-18:
Today I updated the ELRS for my Taranis X9d+ and I couldn't get the Lua script to work with the problem others have described.

I had to update my Taranis to Edge TX 2.8.4 and It solved the problem. Now the Lua script doesn't fail, at least for me.

Kaffiflight, 2023-07-27:
Thanks for the hint, Tried EdgeTX now, sqt5font is not installed. Still does not load, memory issue. I can see the ELRS lua is not big (29) but it´s still the biggest lua script in the collection.
BTW, I can not load the TBS script now, too. Worked before.

MNS26, 2023-08-06:
Kaffiflight would you be willing to test this pull request? ([added float to elrsV3 script](https://github.com/ExpressLRS/ExpressLRS/pull/2345))
It adds support for floats to the ELRS script and allows the ability to access FC settings.
But @pkendall64 is concerned older radios wont be able to run it (like the X9D)
i dont have an old(er) one to test with

Kaffiflight, 2023-08-07:
Yes. I guess I just have to add that to the scripts folder like the version before? I will try.

Kaffiflight, 2023-08-08:
Just for reference: Tried, did not load, did not even start or hang. Memory error at once.

anisoptera, 2023-08-15:
Had this issue as well. Downgrading to 2.5.0 and running the script to generate the .luac, then upgrading to 2.8.5 was successful for me.

xros, 2023-10-23:
> Just for reference: Tried, did not load, did not even start or hang. Memory error at once.

X9D and X9D Plus 2015 have the same chip which has 128KB mem. X9D Plus 2019 has more mem. They are different. I downgraded to version 2.5.0/2.4.0 and `elrsV3.lua` works. But you can't change the baud rate to ELRS transmitter. And it can only work with ELRS transmitter at packet rate 333Hz. It can't reach higher than 500Hz. Nevertherless D250, D500, F500, F1000. 

My another radio radiomaster tx16s mark 2 has no issue. It can run an elrs gemini transmitter packet rate at F1000 (1000Hz), baud rate 5.25Mhz, dual 1W transmitting power (2W in total, 2 antennas)...

paulo-serrao, 2023-10-23:
@xros you can run 500Hz with lua working as long as you don't update past ELRS v3.2.1
on ELRS v3.3 and above you can no longer do 500Hz with the old X9D (lua hangs)

xros, 2023-10-25:
@paulo-serrao My TX is ELRS Gemini which is born with initial version higher than v3.3.0. Everything is cool with my TX16S mark 2, it's just X9D Plus 2015 can only work with them at packet rate 333Hz(full mode) maximumly using EdgeTX 2.5, elrsV3.lua, ELRS 3.3.0 for both TX and RX (all gemini mode). I'm okay with those. The MCU STM32F2 on X9D plus 2015 is really slow.

closed 2024-04-06 by pkendall64

bypbop, 2025-07-22:
Hi,
I try to install ELRS V3 on my taranis x9d Plus. Im on EdgeTX 2.11.2 the last version
Have the same problems "Not enough memory".
Someone can tells me how i can do ?
Regards

bypbop, 2025-07-22:
Thanks for Ur Reply !!
I try the both with the original one work on the simulator (loading) and i copy on SD the luac ! On the radio same not enough memory.
I try alors with the file delete lua and luac from radio i generate with the simulator and have this : version mismatch in  precompileg chunk.
[http://kmeg.free.fr/Capture.JPG](url)

Regards

mha1, 2025-07-22:
Do the simulator thing again.

Copy the simulator generated .luac to the SD card and delete the .lua from the SD card.

If this doesn't work rename the .luac to .lua

bypbop, 2025-07-22:
@mha1  Thanks for ur reply !
Don't have any .luac created when i launch te simulator i launch many times and try to close eTX and restart and try again !
Same ...

mha1, 2025-07-22:
Don't use the file I sent. This us already a compiled lua (luac) but compiled with EdgeTX 2.10. That's probably the reason why you saw the version mismatch.

Just use the elrsV3.lua from the ExpressLRS releases page

NielsDav, 2025-08-12:
> Final attempt. Copy both files contained in the attached zip file to your SCRIPTS/TOOLS folder
> 
> [elrsV3.zip](https://github.com/user-attachments/files/21382469/elrsV3.zip)

I can confirm this works, thanks!

TheSpiderDNA, 2025-09-22:
> As some one mentioned above, deleting touch screen, color screen support and simulator in elrsV3.lua was enough to get it running on my QX7

Could you tell us how to remove support for touchscreen and color screen, etc. on the .lua file please?