ExpressLRS/ExpressLRS #2056: Radiomaster Boxer - 3.2.0

Kendo1985, 2023-01-26:
## Current Behavior

Having a right nightmare.
Anyone know why the boxer freezes when I go into the elrs.
It was working on the latest firmware for a short while and now all of a sudden it gets stuck here on the loading screen.
Video is attached
Thanks

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

Kendo1985, 2023-01-26:
https://user-images.githubusercontent.com/122646718/214732393-19ef396b-3cbe-47e0-9528-1e0649c9add5.MOV

pkendall64, 2023-01-26:
The best place to ask for help is on discord in the #help-and-support channel. You will get a much faster resolution there rather back and forth in a github issue.

Kendo1985, 2023-01-26:
> back

No joy as of yet, been on the discord for about 6 hours straight trying to resolve.

SunjunKim, 2023-01-26:
This happens when the radio is not communicating with the internal module properly.

If the module is soft-bricked, you can try EdgeTX passthrough update (which should work always).
If you've configured everything right and it's still happening, it might be a hardware issue.

534002646, 2023-03-02:
I have the same problem as you, did you solve it?

Kendo1985, 2023-03-04:
> I have the same problem as you, did you solve it?

Yes. Open Model Menu (MDL key)
press Page > key once
scroll down to Internal RF, change it from Off to CRSF. Let me know if it works.

closed 2023-03-19 by brandonrc

GioMBG, 2023-07-26:
Hi Kendo1985,
after going around for hours You resolved my loading... stuck
m a n y - t h a n k s !